mavlink/qgroundcontrol #6345: Detected Radio version - popup inconsistency

ryanjAA, 2018-04-13:
Im replaying some telem logs from a flight last week on PX4 1.73 stable and yesterdays QGC daily. Flight was with a foam plane and some generic SIK telem radios and the log replay is telling me via QGC popup that the “Detected Radio still using mavlink 1.0 on a link with Mavlink 2.0 enabled. Please upgrade the the radio firmware.” It didn't tell me that at the time, only during replay. Haven't changed anything so wondering if this is an actual error or if should be more prevalent during use.

[2018-03-27 16-59-02.tlog.zip](https://github.com/mavlink/qgroundcontrol/files/1908299/2018-03-27.16-59-02.tlog.zip)

[image]

DonLakeFlyer, 2018-04-16:
Have you upgraded the firmware on the radios? Only newer readio firmware supports mavlink 2.

ryanjAA, 2018-04-16:
50/50 on the actual version. I’ll have to track back to the specific radios that were used but besides the version aspect of it. What is the concern is that I only saw this during the flight replay and not during the actual flight.

DonLakeFlyer, 2018-04-16:
Were you using the same version of QGC for both? That check is new.

edsellar, 2018-04-16:
That makes sense now. Let's close this. I was using 3.3.0 stable to fly and was replaying this with v331 and daily from when i posted this.

DonLakeFlyer, 2018-04-16:
Yup. I think that's it. Which would indicate you radio firmware is old. You should re-flash them,.

closed 2018-04-16 by DonLakeFlyer

ryanjAA, 2018-04-16:
Thanks Ed - exactly.

ryanjAA, 2018-04-16:
Yep - will get them re-flashed.