mavlink/qgroundcontrol #6352: Offline Terrain: Crashing on Windows

DonLakeFlyer, 2018-04-14:
@birchera @dogmaphobic Will need your help on this one. This shows up on Windows because the airmap queries are https and we don't install OpenSSL on windows. That said, it exposes a code path which is crashing. No idea how to fix.

The problem starts here: https://github.com/mavlink/qgroundcontrol/blob/master/src/QtLocationPlugin/QGeoMapReplyQGC.cpp#L156. It spits out: ```Fetch tile error: "Error creating SSL context ()"```.

It will then make it to here: https://github.com/mavlink/qgroundcontrol/blob/master/src/Terrain/TerrainQuery.cc#L442. Which is incorrect. The tile should have errored out. Instead it builds a tile with 0 bytes of data: ```Received some bytes of terrain data:  0```.

Which then eventually crashes here: https://github.com/mavlink/qgroundcontrol/blob/master/src/TerrainTile.cc#L122.  I'm assuming because _data is some crazy incorrect size or something like that caused by the bad tile build.

I see two bugs here:
* Why didn't the tile error out since it got a network error?
* This code here: https://github.com/mavlink/qgroundcontrol/blob/master/src/TerrainTile.cc#L47 needs more sanity checking on the input data.

DonLakeFlyer, 2018-04-14:
@antiheavy Here is your crash.

DonLakeFlyer, 2018-04-14:
FYI: I'll figure out how to deal with the OpenSSL thing. I just needs the bugs fixed.

birchera, 2018-04-15:
> FYI: I'll figure out how to deal with the OpenSSL thing. I just needs the bugs fixed.

I think the libeay32 is outdated, we had that on one specific computer earlier. I can supply that fix later today.

I'll also propose some data sanity check for the terrain tiles.

Antiheavy, 2018-04-15:
for what it is worth, i can only get this crash to occur when a fixed wing landing pattern is in the mission plan.  I've tried and tried and cannot reproduce the crash without the fixed wing landing pattern.  With a fixed wing landing pattern the crash sometimes happens quickly, sometimes after many minutes, but it always happens.  Windows 10.

DonLakeFlyer, 2018-04-15:
Ok I see the pulls for openssl and the tile validation stuff. @dogmaphobic What about the fact that the error doesn't propogate correctly?

DonLakeFlyer, 2018-04-15:
I don't want to merge those other pulls until the network reply thing is figured out otherwise it will hide the problem.

dogmaphobic, 2018-04-16:
> What about the fact that the error doesn't propogate correctly?

I've been out most of the weekend and I have not been able to take a look at this. It seems there might be an error in how `QGeoTiledMapReplyQGC` is being used (directly). It was not meant to do what `TerrainTileManager` is doing. Nothing inherently wrong, but because `QGeoTiledMapReplyQGC` was designed to handle only requests from `QGeoTileFetcher`, the signals and responses sent/given by `QGeoTiledMapReplyQGC` may not be enough for the way `TerrainTileManager` works. Everything is fine when there is no error but in case on an error, I don't think there is proper error checking/signaling going on. I need to spend some time (tomorrow) following the flow of the code to see if I can identify these issues.

dogmaphobic, 2018-04-16:
I found some issues and deal with them within #6358. The Json parsing tests as well as the SSL fix for Windows are unrelated to those changes but should take it in consideration.

DonLakeFlyer, 2018-04-16:
Ok, I tested with all the pulls in place. Terrain now works in QGC and doesn't crash. Only problem is the lack of qWarning I noted in Gus' pull which seems wrong.

closed 2018-04-16 by DonLakeFlyer