mavlink/qgroundcontrol #6357: QGC could not save downloaded file to tmp

FaboNo, 2018-04-16:
When posting bug reports, include the following informaiton:
- Ground station operating system ubuntu 16.04
- QGroundControl version V3.2.7
- Autopilot board: Pixhawk 
- Autopilot firmware px4fmu-v4_default.px4
- When I want to update the firmware I got the following error "could not save file on /tmp" (see screenshot below). It is the first time it happens. Actually last friday, I tried to update the firmware but the pixracer was stuck (boatloader issue?) So today I connect with Mission Planner in order to reset the pixracer, but APM was installed per default. And then when I connected with QGC, I got this error.
If somebody can give me a hint? Thank you very much 
Fabrice

[image]

DonLakeFlyer, 2018-04-16:
Can you translate "est un dossier"?

DonLakeFlyer, 2018-04-30:
Also just seeing now you are running 3.2.7. You need to running a newer build Stable or Daily. I believe the loactions to ArduPilot firmware has changed since then. Reopen if new build doesn't work for you.

closed 2018-04-30 by DonLakeFlyer