mavlink/qgroundcontrol #6363: Latest daily doesnt connect to SITL

sanderux, 2018-04-17:
On Ubuntu 16.04

dogmaphobic, 2018-04-17:
I'll be looking at this today...

dogmaphobic, 2018-04-17:
I've been at this for half a day by now and I still can't get this stupid Windows to work. First was getting Qt happy with VS 2017, which apparently installs everything in different paths and Qt can't find anything. Now, after finally having everything built and running, I can't get Gazebo broadcasts to reach QGC on this Windows machine. Firewall rules seem to be correct (got rid of it wholesale) but still nothing...

dogmaphobic, 2018-04-18:
After a zillion experiments, I've isolated the issue to the version of Qt. It stopped working with Qt 5.9.4. If you build with an earlier version, it all works. Now the hunt for the reason...

dogmaphobic, 2018-04-18:
This fixes it: https://github.com/mavlink/qgroundcontrol/pull/6366

dogmaphobic, 2018-04-18:
For completeness sake, when going through all iterations in trying to identify when it stopped working, when I tested it with Qt 5.9.10 (and it worked), I was also running (by mistake as I was trying to keep it methodical) a commit from a couple of months ago. The bug that stopped UDP for all platforms is recent. Some NULL pointer check (that would be impossible for it to ever be NULL) was inverted and not sending anything. Ever. I'm not sure about the Windows only issue. That code has been there since February 5th. Either Qt > 5.9.1 got broken or some new Windows update messed it up.

dagar, 2018-04-18:
Do we need to be tighter with the exact Qt version on all platforms?

dogmaphobic, 2018-04-18:
> Do we need to be tighter with the exact Qt version on all platforms?

Not normally. It's just that there has been a lot of issues with Qt and Windows lately. At the moment, you must use 5.9.4 or greater. We have to wait for Don to be back but we should do a wholesale switch to something more current than what we're using.

DonLakeFlyer, 2018-04-29:
I believe Gus fixed this. Can you retry?

closed 2018-04-30 by DonLakeFlyer