mavlink/qgroundcontrol #6367: gamepad wishlist

dagar, 2018-04-18:
- throttle allow mapping full range [-1, 1] (cars, etc)
 - ideally VTOL transition should toggle from one state to the other
 - arm/disarm require at least 1 second
 - show thumbsticks in the Joystick tab
   - this is minor, but would give the user a better idea about what's going on and provide some necessary configuration (how the throttle is handled)

More advanced configuration
 - allow mapping throttle to other buttons as a +-% change, which frees up the stick for gimbal use

JonReacher, 2018-04-20:
I like the separate switches for VTOL and FW transitions. With the higher latency of the gamepad connection it's nice to have an 'oh crap!' button (back to MC mode) that can be tapped a few times without switching back and forth each time. Maybe it could be an optional choice.

Also, it would be nice to have a way to block arming in FW mode. As it is now in SITL at least, if the button to transition to FW mode is pressed (or bumped accidently) without the operator noticing on the screen, arming the system with center throttle (full down is zero option on centered gamepad sticks) will send the forward motor into half throttle instantly. Not very safe, so I hesitate to use an arm button, although it would be very nice to use one.

I second the % throttle change button mode. It would be great for position mode. Gimbal sticks on a gamepad without a second operator would be very nice.

dagar, 2018-04-20:
- don't allow thumbsticks + joystick simultaneously (https://github.com/PX4/Firmware/issues/9346)

closed 2026-10-05 by github-actions[bot]