mavlink/qgroundcontrol #6369: suggestions for a more user-friendly QGC user interface

barzanisar, 2018-04-19:
Hi,

I wanted to suggest a few improvements for QGC user interface. It would be nice to show the names of the main views in the toolbar whenever the user hovers his/her mouse over it.
[image]

Also, for users new to QGC, it would be nice to ask them if they want a quick start tutorial of QGC, the first time they use it. If they choose yes, the interactive tutorial should navigate through the main views explaining in text boxes what each main view is used for and its main features. Just how when we install a new app in smartphones and it gives a quickstart tutorial which we can also opt to skip. This tutorial can also be set as an option in the Application settings, which the user can access later on as well if they forget anything. 

Another example of the tutorial design could be that whenever user hovers over an option/menu a text box should appear explaining the option in a couple of lines.

@thomasgubler

thomasgubler, 2018-04-20:
@LorenzMeier @dogmaphobic looking for your input on the suggestions by Barza

Antiheavy, 2018-04-20:
All the features @barzanisar describes above would be awesome. 

Mouse hover descriptions work great when using a mouse, but don't work really with touch screen interfaces - i don't know if the world has solved the problem yet or not.  (Having said that I always use a mouse anyway because QGC touch interface in Windows isn't really usable.)

DonLakeFlyer, 2018-04-29:
> Mouse hover descriptions work great when using a mouse, but don't work really with touch screen interfaces - i don't know if the world has solved the problem yet or not.

Which is why they haven't been added. Also there were previously massive problems/bug with Qt tooltip style support so adding them for desktop builds only was a problem as well. Maybe they work now in Qt, don't know. Those problems were years ago.

Instead each view has it's name in the upper left on the toolstrip so at least you can click through and get names that relate to documentation.

@barzanisar For the visual cut-off problems you describe can you enter separate issues for those. They are just bugs which should be fixed.

DonLakeFlyer, 2018-04-29:
> Having said that I always use a mouse anyway because QGC touch interface in Windows isn't really usable.

Is that because the touch targets are too small? There needs to be a setting on desktop for expanding touch targets just like mobile does.

Antiheavy, 2018-04-30:
> Is that because the touch targets are too small?

I think that might be part of the problem.  I know a few of us who've tried using touchscreen window's machines (Microsoft Surface Book and a cheapo "netbook"), we all quickly gave up and went back to KB/touchpad.  One of these day's we'll experiment more and post separate issue.

DonLakeFlyer, 2018-04-30:
Ok, I just get a touchscreen windows machine so I'll give it a try to see how it works.

DonLakeFlyer, 2018-04-30:
Not trying to downplay usability problems. Just trying to bring everyone up to date on things that were already tried or not tried. For example the old QWidget ui had tooltips on many things. But they also got to be really annoying (a lot of it caused by shitty Qt implemenation) and were not brought across to Qml ui changeover. Also when QGC switched to Qml the ui focus shifted to more of a tablet/touch centric model which doesn't have a good standard tooltip solution replacement. Hence tooltips were never though of as a good solution. And due to that desktop/mouse kind of got the short end of the straw.

dogmaphobic, 2020-02-11:
Ancient history

closed 2020-02-11 by dogmaphobic