mavlink/qgroundcontrol #6371: Questions related to Home Position in Mission mode

barzanisar, 2018-04-20:
Question 1) What is the purpose of Home position? Is it just so that the vehicle knows where to go back when RTL command is given in mission mode?

Question 2)
I am using Pixracer to fly missions in HITL. By default, the home position is set on the vehicle. However, if I want to change the home position I can EITHER change map center and choose "Set Home To Map Center" under "Planned Home Position" OR click on home waypoint and drag/drop to another location:
[image]

But this does not change my home position because after I upload the above plan and switch to Fly view, the home position still shows on the vehicle.
[image]

When I start this mission, the vehicle returns to the same home position shown above and not what I "planned" as in the first picture.

How can I change my home position to be not set on vehicle?

Question 3) What is the difference between the "Planned Home Position" and "Actual Home Position" ?

Question 4) What is the purpose of "Home Position" command? I added this as waypoint 3 but then uploading mission fails, with the error message as shown below:
[image]

@dagar

Antiheavy, 2018-04-21:
@barzanisar QGC and PX4 don't really talk to each other about Home position.  I guess there isn't really a MAVlink protocol for that exchange.  It is too bad, because there really should be better handshaking between PX4 and QGC on this.  with this in mind I'll take a stab and your questions:

> Question 1) What is the purpose of Home position?

The autopilot's home position is used for a bunch of things, primarily it is used as a location to Return to when in Return mode.  PX4 periodically updates it's home position to be the location of the vehicle whenever it is not flying.  This is most noticeable when you first get a good GPS position, when you takeoff, and after you land.  The PX4 actual Home position is displayed in the Fly screen on QGC.

> Question 2)

> How can I change my home position to be not set on vehicle?

you can't.  PX4 sets it's own home position automatically.  This is actually good, safe behavior.  What you (and me and everyone else in the world) find confusing is that the Home position in the QGC Plan screen has nothing to do with the actual Home position of the autopilot.  It is just shown there for planning purposes, in particular when planning without being connected to a vehicle with its own Home position.

> Question 3) What is the difference between the "Planned Home Position" and "Actual Home Position"? It seems like from QGC doc planned home position just has visual significance and is not really where the vehicle will return to when we RTL in the end of the mission. Whereas, actual home position is the first position where vehicle arms after uploading the mission, regardless of the flight mode we armed the vehicle in. Am I right? Can someone please confirm this?

yeah, you are basically right I think.  Except the actual Home position is periodically updated by the autopilot, including when you arm.

> Question 4) What is the purpose of "Home Position" command? I added this as waypoint 3 but then uploading mission fails,

interesting.  I wonder if this is a command that is supported by other autopilot firmwares other than PX4?  I've never used that.  Personally I am hopeful QGC will eventually get smarter at filtering waypoint types and mission commands that are not available for a given vehicle (lots of work to get there I think).

dagar, 2018-04-21:
QGC and PX4 have some communication in place about the home position, but there are still some weird cases in QGC where it shows a home (H) that's not what the vehicle has. 

We have most of the required pieces in place for both QGC and PX4 to support the planned HOME position with only a few small changes. Someone just needs to go through it from the top and define the interaction.

 - when would you want to manually set a home position?
 - what's the proper interface for manually setting a home position?
  - mavlink has SET_HOME_POSITION (https://mavlink.io/en/messages/common.html#SET_HOME_POSITION), but also allows you to insert a HOME position as a COMMAND in a mission. https://mavlink.io/en/messages/common.html#MAV_CMD_DO_SET_HOME

I could almost see wanting to keep both, but this really needs to be driven by actual requirements.
For example the vehicle could maintain the concept of always recording where it last took off from, but also allow a specific mission to change HOME part way through? Outside of a planned mission I could see manually setting home as a useful feature for testing or planning a mission before you have GPS.

Let me know if there's interest in pursuing this. At a minimum we should update QGC to not show a home position if it didn't come from the vehicle recently.

Antiheavy, 2018-04-21:
related: https://github.com/mavlink/qgroundcontrol/issues/5446

barzanisar, 2018-04-23:
@dagar As I understand, whenever I change Home position in Plan view either by choosing **Set Home To Map Center** under "Planned Home Position" OR by dragging the home waypoint around, this only changes the "Planned" Home position shown on QGC, which is just meant for visualization of mission on QGC. However, when I go to the airfield and start the mission, the autopilot takes its takeoff position as its actual home position. 

The only reason why I personally would want to manually change the actual home position is: if I want to return to some other position (instead of the takeoff position) in Return mode. This can alternatively be achieved by simply adding another waypoint and land there. So I don't see any reason why someone would want to manually change actual home position from QGC.

Correct me if I have understood you wrongly: 
The only way to manually change the actual home position (the position vehicle travels to and lands) from QGC is to add a waypoint "Home Position" that corresponds to the command "MAV_CMD_DO_SET_HOME". Is this correct?

My questions are just intended to improve the QGC documentation, so that it is clear to the user the difference between Actual vs Planned Home Position and if the Actual Home can be changed from QGC.

barzanisar, 2018-04-29:
@dogmaphobic @DonLakeFlyer  Can anyone of you please clear my confusion mentioned as the comment above so that I can commit to (this PR)[https://github.com/mavlink/qgc-user-guide/pull/137].

DonLakeFlyer, 2018-04-29:
The fact that you were able to change a command to "Home Position" is a bug. This is not the mavlink DO_SET_HOME command to set a home position in a mission. This is a bug around how "Planned Home Position" is represented internally to QGC causing it to show up wrongly as an item you can change to. It is actually a command which does not exist and should never be shown.

QGC only shows you commands which are implemented by the firmware type you are using. This firmware type either comes from the connected vehicle or if you are planning offline you can set the firmware type in Vehicle Info section of Mission Start. QGC does not currently filter out commands by vehicle type. So for example you will see VTOL commands supported by the firmware being available even if your vehicle is not VTOL. If the firmware type is set to Mavlink Generic all commands are shown since in that case we do not know which commands are not supported.

Neither PX4 nor ArduPilot support DO_SET_HOME hence it will not show up as being available to change to. Just for FYI this command would show up as "Set home location" with a description of 
"Changes the home location either to the current location or a specified location.". If PX4 starts to support DO_SET_HOME it can be added to the supported commands list for PX4 firmware.

Does that answer all the questions?

NaterGator, 2018-04-29:
But PX4 does support DO_SET_HOME: https://github.com/PX4/Firmware/blob/fbbe1f5288f0e3255b083f8deaaaf2168202a16a/src/modules/commander/commander.cpp#L847

DonLakeFlyer, 2018-04-29:
Then QGC is just out of date. I'll add it to supported commands.

NaterGator, 2018-04-29:
The caveat here is (if my memory serves) PX4 will automatically replace the home position as others specified during certain conditions (arming, GPS lock, takeoff). This behavior makes it kinda unintuitive to predict where home is if you manually set it with DO_SET_HOME and it subsequently gets replaced. 

I think input and changes are probably needed on both ends to make the behavior predictable to track what is most desirable given the level of user engagement. (IE: if they manually configured home position, maybe don't automatically override it?). Maybe things have already changed on that front since I last used it.

DonLakeFlyer, 2018-04-29:
That would all seem like firmware side changes though. I'll add a firmware issue which links to this and can continue discussion there.

DonLakeFlyer, 2018-04-29:
Also just FYI that I understand that the "Planned Home Position" thing is a major ball of somewhat not understood crap. The goal is to be able to plan a mission while not at the field connected to a vehicle and still be able to know things like mission flight time, battery count, ... as well as the visual path of the mission. If anyone has better ideas as to how to make this understandable I'm all ears.

hamishwillee, 2018-04-29:
> If anyone has better ideas as to how to make this understandable I'm all ears.

Something that at least makes the home icon visually different when not connected. E.g. vehicle icon and path to first point semi-transparent/faded. An overlay on the icon: "planned".

Antiheavy, 2018-04-30:
Hamish’s suggestion is a decent idea that would reduce confusion between the Home point on the Plan screen (not actual home) vs the Home point on the Fly screen (vehicle sets this point and tells QGC).

DonLakeFlyer, 2018-04-30:
The

> The fact that you were able to change a command to "Home Position" is a bug. This is not the mavlink DO_SET_HOME command to set a home position in a mission. This is a bug around how "Planned Home [...]

This is fixed now.

closed 2018-04-30 by DonLakeFlyer