mavlink/qgroundcontrol #6375: Why mobile has no Mavlink Console Page ？

MyOnlyOnlySunshine, 2018-04-25:
[image]

Hello,I use qgroundcontrol on iPad, but I haven't found Mavlink Console Page on iPad.Then i check the source code ,it doesn't built MavlinkConsoleController on iPad,Coule anyone tell me why mobile doesn't contains that page. I want to check gps status on that page. Thanks.

DonLakeFlyer, 2018-04-29:
@NaterGator Can't quite remember the reason for this?

DonLakeFlyer, 2018-04-29:
@dogmaphobic Or is it because Mobile just tosses whole Analyze view?

dogmaphobic, 2018-04-30:
I don't remember ever touching that code. It seems the entire Analyze view is disabled on mobile. Geotagging doesn't seem it makes sense on mobile. Log Download has dubious benefits on mobile as well. I guess MAVLink console, which was added way later got stuck in a menu that is only present on desktop builds.

DonLakeFlyer, 2018-04-30:
Yeah I think that's pretty much it.

DonLakeFlyer, 2018-04-30:
Dup of #6133

closed 2024-06-02 by HTRamsey