mavlink/qgroundcontrol #6386: Test QGC on a Windows touch screen machine

DonLakeFlyer, 2018-04-30:
Reports of it being crap. I'm thinking because touch areas are not expanded when running desktop build with a touchscreen.

b14ckyy, 2018-05-05:
Hi Don. I just moved to Pixhawk and Ardupilot (because iNAV totally stucks in development) and QGround Control immediately become my No.1 GCS. Using it on Notebook and Windows 10 Tablet. So I hope I can give a few useful tips. 

Hardware: TrekStor Surf Tab W1 3G / Intel Atom Z830 / Windows 10 1803 (x32) / 2GB RAM / Cheapest of the cheapest crap.

Main issue is performance of Map loading. When loading buffered or pre-loaded Map data, on slow devices it takes forever until the application gets responsive. So it is better to not pre-load maps and load over 3G instead and only use cache buffered maps.

The UI in general is very touch friendly in my opinion. Just need to scale Fonts up to 12 to get nice button size on an 1080p 10 Inch screen. Could it be possible to auto-size fonts after first install, depending on Windows 10 Display Scaling level? Display Scaling is 150% on my device and with 150% Font size it is perfect. Or rename Font Size to GUI Scale because it does scale up everything anyway?

Pan and Zoom in map is a bit sluggish. Performance Optimizations here would be nice but I could live with that if not possible. 

Value Widget Headline, Settings Button and Dropdown does not scale up and is hard to hit with finger input.

Status Line for data sync (below the status bar) should be better visible and also scale up a bit. It is very thin and barely seeable outdoors in daylight.

Editing number values on waypoints is annoying. There should be a trigger that opens On Screen Keyboard when tapping inside a text field. Or even better > Open a custom Number-Block than can be activated in settings when using tablet.

The Waypoint Menu dropdown (Hamburger menu in waypoint window) should also be scaled up. It is hard t delete a waypoint. Or you could add additional buttons beside this hamburger menu for most common actions (Insert Waypoint, Delete, Insert Pattern).

Log view is also not scaled up and it is hard to select specific logs for download. This should also be scaled up. 

These are my points for now. Hope this helps. If I should try something more, just ask :)

DonLakeFlyer, 2018-05-06:
Thanks a ton for all the details. Question on this: 

> Main issue is performance of Map loading. When loading buffered or pre-loaded Map data, on slow devices it takes forever until the application gets responsive. So it is better to not pre-load maps and load over 3G instead and only use cache buffered maps.

I don't quite get what you are saying here. Are you saying:
* Specifically using Offline Maps and caching doesn't help perf?
* Downloading tiles on the fly over the internet by just panning around/zooming the map is bad?

DonLakeFlyer, 2018-05-06:
The fundamental problem with using Windows and a touchscreen which is a tablet style device is that QGC still runs this like a desktop build. On Mobile builds which all assume touch the UI changes/scales to provide larger touch targets. As well as automatically providing virtual keyboard for entering values. There needs to be a way to do the same thing on what are essentially desktop devices which are touch screens.

b14ckyy, 2018-05-20:
> Specifically using Offline Maps and caching doesn't help perf?
Exactly. Loading the cache when starting the application takes a lot of disk Performance and for a 500MB cached map I have to wait up to 2 minutes until the application can be used. 

> Downloading tiles on the fly over the internet by just panning around/zooming the map is bad?
In General that is fine but panning and zooming takes a lot of Performance. So on slow Systems and tablets, it stucks a lot.

DonLakeFlyer, 2018-05-20:
> Loading the cache when starting the application takes a lot of disk Performance and for a 500MB cached map I have to wait up to 2 minutes until the application can be used.

That would indicate something is wrong. It doesn't matter how big the offline cache is. It just pulls the tiles it needs. It hangs for 2 minutes when? At boot?

DonLakeFlyer, 2018-05-21:
@dogmaphobic This seems strange. Any ideas?

appilivamsikrishna, 2018-06-28:
QGC doesn't work on windows 10 Home Version 1803, OS built 17134.81 .. The problem here is it is not at all opening GUI., when I click on the program starter . I have tried opening the program in safe mode also. I tried opening in all modes. it doesn't give any clue. 
Please Help …. 

V 2.9.5 Works well in this device.. Please help

b14ckyy, 2018-07-01:
You should open a new issue for that, this ticket is for Touch optimization.

b14ckyy, 2018-07-01:
@DonLakeFlyer 
I have a temporary Lenovo Idea Pad at the Moment with only 1280x800 sceeen Resolution but with 4 gig Memory, double the amount than my TrekStor. 

I'll test the cached map loading behavior again with latest daily

b14ckyy, 2018-07-01:
Finally tested it on the Ideapad. Version now is Build 0700 from 2018-06-29.

Did you change anything about performance already in 3.3.2? It runs muuuuch better now. It does not use GPU resources anymore when map is still standing and CPU load is 0%. So it does not drain my mobile device battery as excessive as before. Also now with about 1GB of preloaded hybrid map data, launch time seems not to be affected anymore.

Pinch to zoom performance is definitely better on an Atom Z8350 CPU, compared to 3.3.2. It does not stuck the whole time and is much smoother and usable. 

The ROI distance is not calculated into flight distance anymore, that's nice too. 

What about my Idea of generate such a ROI Delete waypoint instead of setting values to zero, when selecting Arducopter as Firmware?

Btw the Waypoint category selector dropdown should become a bit finger frendlier. It stays very small atm even with font size set to 12px to have nice large buttons.