mavlink/qgroundcontrol #6387: *Daily build* - Rel.Abs.AGL. in Waypoint editor lose sync

Antiheavy, 2018-04-30:
testing Daily build 2018-04-29

three screenshots with the AGL, Abs, Rel radio selectors, notice the altitude preview pane in the lower left.  There is no change between AGL and Abs (should be a very noticable change) and there is a strangely  large change between Abs and Rel. (wrongly).  Seems to depend on which order you click/switch between the different altitude modes.
[image]
[image]
[image]

DonLakeFlyer, 2018-04-30:
Ah, I knew I'd miss something. This is all new, like 10 minutes old new! Some signalling is missing somewhere.

DonLakeFlyer, 2018-04-30:
Fixed

closed 2018-04-30 by DonLakeFlyer