mavlink/qgroundcontrol #6391: Terrain info doesn’t work with English units

Antiheavy, 2018-04-30:
Terrain info doesn’t work and other bugs (upload button no flash) when using English units.   (Got something against the size of the kings shoes?)

DonLakeFlyer, 2018-04-30:
I think I may have just fixed that here: https://github.com/mavlink/qgroundcontrol/pull/6389. The home altitude was using the wrong units for altitude. Not sure if new daily with that is out yet.

DonLakeFlyer, 2018-04-30:
Are you talking about the terrain height display at the bottom shows wrong heights for waypoints? Or something else?

Antiheavy, 2018-04-30:
Shows no height for waypoints in the lower left at all.  and also the Upload Required button disappears without clicking on it to Upload.

Antiheavy, 2018-04-30:
Works fine with meters. Nothing with feet.

DonLakeFlyer, 2018-04-30:
Can you try again with latest daily? It's all working for me now. Still can't repro the upload button thing though.

Antiheavy, 2018-05-01:
Now I'm trying it from a hotel and I can't get any terrain to work at all, regardless of meters or feet units.  I've tried both the hotel Wi-Fi (pretty terrible) and tethered to my phone's LTE (better).   Trying both the daily from a few days ago and today's 2018-04-30 daily.  same results which look like this in the console:
[image]

DonLakeFlyer, 2018-05-01:
Looks like the whole internet connection is bonkers since map tiles aren't downloading either.

Antiheavy, 2018-05-02:
working in both feet and meters now with the latest daily.  I'm not sure what is wrong with my internet connection to airmaps on the laptop I am using, very flaky.  tried on another machine and it seems to work correctly.

closed 2018-05-02 by Antiheavy