mavlink/qgroundcontrol #6413: QGroundControl not reacting to heartbeat message

simonbeaudoin0935, 2018-05-03:
Hi,

I have found nowhere information about my current issue, so i decided to open an isseue to describe it.

Last summer i was playing with qgroundcontrol with a raspberry pi and i wrote a program that runs on one and used mavlink to talk to qgc. When i downloaded the latest release (3.3.2) today, and tried to use my old program, qgc never recognized the heartbeat messages and never showed up a new device, i tried using UDP and serial, no luck. Furthermore i compiled the little linux test program just to convince myself that the error could not be from my side (https://github.com/mavlink/mavlink/blob/master/examples/linux/mavlink_udp.c) which basically does the same as my program, but no luck. I do see in the terminal that once i start sending messages i begin receiving heartbeat, but only heartbeat (it then means that my own heatbeats dont reach qgc properly, otherwise i would have received other messages such as param request list, etc). 

I then looked at the date of creation of my old project, wich is around may 2017 and i went and downloaded QGC 3.1.3 that was released before but not after i began to play with the software. I then works flawless. Tries the version right after, 3.2.0, then nothing. The same test program is used in all cases. 

Hope that someone can explain me something i did't get !

DonLakeFlyer, 2018-05-18:
Here is how QGC determines if there is a vehicle on the link: https://github.com/mavlink/qgroundcontrol/blob/master/src/Vehicle/MultiVehicleManager.cc#L75

closed 2018-05-18 by DonLakeFlyer

Desync-o-tron, 2019-04-05:
@DonLakeFlyer Thanks a ton for pointing this out. 
I've been trying to get the DJI M100 DJI2MAV ROS node to connect to QGC for 20+ hours now. I commented out lines 85-97 on the file you linked above, and now the drone connects & flies in a HITL simulation, where before I couldn't get a vehicle connection.