mavlink/qgroundcontrol #6419: Yaw control not working, copter spins while in flight

JoshLikesBeer, 2018-05-04:
I have a GameSir T1 controller paired with an Android 7.0 tablet running the current version of QGroundControl from the Play Store.

On the Joystick calibration page, the Roll, Pitch, and Throttle controls all work as expected. The roll and pitch are at the center when the right stick is centered, the throttle is at the bottom while the left stick is centered, and these controls all move in sync with the Axis Monitor.

The yaw control is a different story. When the left stick is centered, the yaw is either all the way over to the right (usually) or all the way over to the left (sometimes). It doesn't respond very well to the left-right motion of the left stick. The Axis Monitor correctly reflects the motion of the stick.

This is in mode 2. In other modes, whichever control is mapped to the left-right motion of the left stick has this behavior.

Here is a video if it's not clear what I'm trying to say.
https://youtu.be/wfQN_ewZ4H8

JoshLikesBeer, 2018-05-07:
I just tried the daily build instead of the stable build, but I still have almost the same behavior. The yaw control is still not reacting correctly to the left-right motion of the left stick, while the axis monitor continues to correctly reflect the motion of the stick.

The difference I see between the two versions is that in the stable version I was able to intermittently move the yaw control to the left by fiddling with the left stick, but in the daily build the yaw control stays locked to the right and I can't finish the calibration procedure.

JoshLikesBeer, 2018-05-08:
If it helps, here are screenshots of my summary page.
[image]
[image]

rollysing, 2018-05-08:
- I may be off-base here and may not be related or the table app is different from the PC QGC, but was just curious why the Yaw is Channel 7?
- Also in your video, in the Axis Monitor, 2 sliders are moving with Yaw and Throttle
- The Axis Monitor numbering doesn't seem to correspond to the Channel assignment according to your Radio tab either. i.e.:
     - Roll is on 6 but the tab says it's 1
     - Pitch is on 7 bur the tab says it's 2

Good luck.

JoshLikesBeer, 2018-05-09:
This might also help.  I set SDLOG_MODE="from boot until shutdown" and I set SDLOG_PROFILE=32 (debug), and then rebooted the drone.

I attached the logcat that starts when I launched the QGroundControl daily build. It loaded the parameters, then I went into Settings -> Joystick and attempted to calibrate. I got as far as calibrating the yaw where it gets stuck because I can't make it go left.

[log02.txt.gz](https://github.com/mavlink/qgroundcontrol/files/1986054/log02.txt.gz)

If it's helpful, I can do the same with the released version. Let me know.

DonLakeFlyer, 2018-05-09:
Can you turn on JoyStickLog and JoyStickValuesLog using these instructions: https://docs.qgroundcontrol.com/en/SettingsView/console_logging.html. Then try to calibrate the joystick. Then save the console output file and attach here.

JoshLikesBeer, 2018-05-09:
Here is what I captured. The JoyStickValuesLog was spamming the log like crazy.
[AppLog.txt](https://github.com/mavlink/qgroundcontrol/files/1989607/AppLog.txt)

DonLakeFlyer, 2018-05-10:
Sorry I missed one. Also need: JoystickConfigControllerLog.

JoshLikesBeer, 2018-05-11:
This has all the Joystick logs turned on.
[GameSir.txt.zip](https://github.com/mavlink/qgroundcontrol/files/1995851/GameSir.txt.zip)

DonLakeFlyer, 2018-05-11:
I don't know what capabilities this GameSir T1 controller has. But when you move a stick it reports two axes as moving fully to the max which in turn confuses QGC calibration. Unless there is some way to turn that off you won't be able to calibrate this. QGC just gets confused when it is looking for a single axis to move and more than one axis moves.

JoshLikesBeer, 2018-05-11:
Thanks. I have no idea if that's possible, but I'll look.

Is it possible to tell QGC to ignore one of the axes?

DonLakeFlyer, 2018-05-12:
> Is it possible to tell QGC to ignore one of the axes?

Not possible.

DonLakeFlyer, 2018-05-12:
@jaxxzer Have you guys ever seen anything like this?

JoshLikesBeer, 2018-05-14:
Here's some more information that might be helpful. As you can see in the video, when I try to calibrate this controller in QGroundControl, it shows left-right motion on the left stick simultaneously on the 1 and 5 axes, and shows up-down motion on the left stick simultaneously on the 0 and 4 axes. Motion on the right stick is only shown on one axis for each of left-right and up-down. I have the same behavior on released version and daily build.

I installed [Controller Test (Unity Mapping 28 Axis)](https://play.google.com/store/apps/details?id=com.eklkgo.UnityControllerTest) on this tablet, and it shows left-right motion on the left stick on the 1 axis only, and shows up-down motion on the 2 axis only. Any ideas why QGroundControl is receiving different input from this controller vs other apps?

jaxxzer, 2018-05-14:
This is apparently an issue with the joystick driver. @JoshLikesBeer can you try the same thing on a desktop/laptop?

JoshLikesBeer, 2018-05-14:
I'm trying on my laptop (running Linux Mint 18.3). I paired the controller to the laptop, pointed the drone's qgc.conf to my laptop and restarted the mavlink-router daemon, then ran QGroundControl on the laptop, but I don't have a Joystick tab in QGroundControl.

[image]

closed 2024-06-02 by HTRamsey