mavlink/qgroundcontrol #6421: Issue: Survay Parameter not Loaded from .plan file

b14ckyy, 2018-05-05:
QGC Version: 3.3.2 Stable
OS: Windows 10 1803 x64
Autopilot: Pixhawk 2.4.8, Arducopter 3.5.5

Steps to reproduce:
- Plan Mission
- insert survey
- change any survey parameter (spacing, cam trigger and so on)
- add RTL
- save mission as .plan file
- load .plan file
> all survey parameters reset to default value
- check .plan file in editor > correct values are stored so it seems they are just ignored on load

Screenshots and .plan file added

[Survey Test.zip](https://github.com/mavlink/qgroundcontrol/files/1976459/Survey.Test.zip)

[image]
[image]
[image]

b14ckyy, 2018-05-05:
Just also tested with latest Daily Build and there this seems to be fixed. Loaded same .plan file and all fine here. Will test now and if this version does not cause major issues, I will keep this for now till next release.

DonLakeFlyer, 2018-05-06:
Fixed in Stable 3.3. New code in Daily doesn't have this bug.

Thanks so much for the detailed bug report.

closed 2018-05-06 by DonLakeFlyer