mavlink/qgroundcontrol #6427: Rounded survey corners

dgorissen, 2018-05-06:
Maybe there is an option for this but I havent found it so far.

When flying surveys with a multirotor the drone goes through a lot of acceleration/decelerations as it tries to follow the corners exactly, pretty much coming to a complete standstill on each turn. 

Especially when flying at a reasonably high speed this reduces battery life. An option to connect the survey lines with semi-circles (or similar) instead of hard corners would be nice.

DonLakeFlyer, 2018-05-06:
I believe that is a flight dynamic of the firmware implemenation. There isn't really a generic way to get the vehicle to do the semi-circle. ArduPilot has splines but PX4 does not. But still not sure what thay would fly like. The transistions from one waypoint to the next are all handled by the firmware.

What firmware are you flying?

dgorissen, 2018-05-07:
Mmm thinking about it, you are right. Googling around seems this is the key issue: https://github.com/ArduPilot/ardupilot/issues/6757 and other than fiddling with WPNAV_RADIUS and accel/speed limits looks like there is no direct way to control this.

closed 2018-05-07 by dgorissen