mavlink/qgroundcontrol #6428: NAV_RCL_LT option missing from Safety view

hamishwillee, 2018-05-07:
PX4 docs state that if [NAV_RCL_ACT](https://docs.px4.io/en/advanced_config/parameter_reference.html#NAV_RCL_ACT) is set to `4: RC Auto Recovery (CASA Outback Challenge rules)` then [NAV_RCL_LT](https://docs.px4.io/en/advanced_config/parameter_reference.html#NAV_RCL_LT) can be used to set loiter time.

However the Safety page does not display the additional field even if NAV_RCL_ACT is set to 4.

I don't consider this a priority because Casa rules are not clearly documented *anywhere* - just posting FYI.

----
When posting bug reports, include the following informaiton:
- Ground station operating system: Linux
- QGroundControl version and build (daily, stable, self-built from source, etc.): Downloaded daily build from Friday 4 May 2018
- Autopilot firmware (e.g. PX4, ArduPilot, custom) and version: PX4 master Friday May 4 2018
- Exact steps to reproduce the problem. Starting from booting QGroundControl to when the problem occurs.

DonLakeFlyer, 2018-05-07:
The Outback challenge stuff (there are more params for this) are a set of super power things specific to the outback challenge. When designing the Safety page it was decided in discussion with firmware folks to not include any of it. Power users can still acess from parameter editor if they want it.

hamishwillee, 2018-05-07:
Thanks Don - makes sense

closed 2018-05-07 by hamishwillee