mavlink/qgroundcontrol #6434: Feature Request Custom Widget to read MAVLink Messages

IntricateInk, 2018-05-08:
I am trying extract mavlink messages (106 in common) for optical flow using the CustomCommandWidget through QML. This is currently not available through Vehicle QML properties.

Would it be possible to expose the valueChanged signal from MAVLinkDecoder through QGroundControlQmlGlobal for custom widgets to access?

hamishwillee, 2018-05-08:
Not my place to say, but I suspect not. Have you tried **Widgets > Analyze** or **Widgets > MAVLink Inspector**. These are normally the right way to get real-time mavlink information in QGC. The main difference being that Analyze allows you to plot graphs

IntricateInk, 2018-05-09:
I'm trying to develop something similar to the graph in Widgets > Analyze using the xy optical flow values.

I'm aware that Widgets > Analyze / MAVLink Inspector can obtain the values in real time, but I also have a need to process this data in real time as well. I'm also aware that its possible to obtain the values from the ulog, but that isn't real time either.

To my understanding, the MAVLink Inspector and Linegraph obtain their values from the MAVLinkDecoder signal, which already exists but is not directly accessible through custom widgets.

The QGroundControlQmlGlobal class is already accessible through custom widgets and holds a private reference to MAVLinkDecoder. I'm not sure if I missed some way of obtaining the decoded mavlink values, but being able to access the mavlink data stream would allow for a lot of non-specific functionality to be developed. 

Currently (as far as I'm aware) the custom widget is only capable of obtaining and setting values that are already readily available in the QGC ui, which isn't very helpful for extending its functionality.

DonLakeFlyer, 2018-05-09:
The long term plan is to expose all mavlink traffic through the Fact System. But that is in the future. Until then things are stuck as they are now.

IntricateInk, 2018-05-09:
Appreciate the reply, thank you.

closed 2018-05-09 by IntricateInk