mavlink/qgroundcontrol #6435: ArduPilot: Support for Trad Heli

lvale, 2018-05-09:
Hi Don

Just as a placeholder for starting the support for setting up a ArduPilot TradHeli

ChristopherOlson, 2018-05-09:
Since I will be working on this as well, I'll make a post here so I get notification of comments.

DonLakeFlyer, 2018-05-20:
Questions:
* Is Trad Heli the "Heli" frame of ArduCopter or something different? 
* Does it report MAV_TYPE=MAV_TYPE_HELICOPTER in heartbeat?

ChristopherOlson, 2018-05-20:
Hi Don,
TradHeli was the long time thing it was called, but I think "Heli" is a better term. As we actually have three different frame types - traditional (with tail rotor), tandem and quad. And the term "Heli" covers them all.

MAV_TYPE=MAV_TYPE_HELICOPTER should be correct. That is the standard MavLink message set spec. QGC properly shows the frame class as "heli" so it must be understanding the message set currently used:

[image]

DonLakeFlyer, 2018-06-05:
We need a page with five categories grouped.
Servo Setup with the following settings in it in this order:
SERVO1_FUNCTION
SERVO1_MAX
SERVO1_MIN
SERVO1_REVERSED
SERVO1_TRIM
SERVO2_FUNCTION
SERVO2_MAX
SERVO2_MIN
SERVO2_REVERSED
SERVO2_TRIM
SERVO3_FUNCTION
SERVO3_MAX
SERVO3_MIN
SERVO3_REVERSED
SERVO3_TRIM
SERVO4_FUNCTION
SERVO4_MAX
SERVO4_MIN
SERVO4_REVERSED
SERVO4_TRIM
Swash Setup with the following settings in it in this order:
H_SV_MAN
H_SV1_POS
H_SV2_POS
H_SV3_POS
H_SWASH_TYPE
H_COL_CTRL_DIR
H_FLYBAR_MODE
H_COL_MAX
H_COL_MID
H_COL_MIN
H_CYC_MAX
Throttle Setup with the following settings in it in this order:
H_RSC_MODE
H_RSC_CRITICAL
H_RSC_IDLE
H_RSC_RAMP_TIME
H_RSC_RUNUP_TIME
H_RSC_SETPOINT
H_RSC_THRCRV_0
H_RSC_THRCRV_25
H_RSC_THRCRV_50
H_RSC_THRCRV_75
H_RSC_THRCRV_100
Collective Curve Setup with the following settings in it in this order:
H_LAND_COL_MIN
IM_STAB_COL_1
IM_STAB_COL_2
IM_STAB_COL_3
IM_STAB_COL_4
Tuning Setup with the following settings in it - once we go to new defaults for tuning, these are all that need to be adjusted for different size heli's on numerous helicopters I have tested, from 500's to 800's, that will fly them stable from hover to 100 mph (if they're capable of those speeds).
Note RC_FEEL has been deprecated in 3.6
ATC_ACCEL_P_MAX
ATC_ACCEL_R_MAX
ATC_ACCEL_Y_MAX
ATC_RATE_P_MAX
ATC_RATE_R_MAX
ATC_RATE_Y_MAX
ATC_RAT_PIT_VFF
ATC_RAT_RLL_VFF
ATC_RAT_YAW_VFF
ATC_RAT_YAW_P
ATC_RAT_YAW_D
For each of these groups you can have fancy sliders, knobs or gauges as desired. But they are not necessary. GUI's already co-exist for multi/heli for all the other stuff, from accel/compass cal, to RC cal, failsafes, battery/power module, etc., in QGC
1) Display the human-understandable description and options for each setting instead of the cryptic parameter name and number options. QGC already does this anyway. Only show the cryptic param name if an "Advanced Mode" box is checked
2) Only allow making one setting at a time - do not batch settings. QGC already does this anyway. Only allowing one setting at at time prevents mistakes
3) Make sure the default ranges fit helicopters so QGC does not return an error that the setting is out of range. This comes from the code, so if we find any we'll have to fix it in the code

ChristopherOlson, 2018-06-05:
@DonLakeFlyer I see Luis must've gotten this info to you! This is great. I wish I understood how to code a GUI application so I could help with this!

ChristopherOlson, 2018-06-06:
Well, beauty is in the eyes of the beholder, and having groupings of the settings like that is what I had in mind. Will this build for Android as well?

DonLakeFlyer, 2018-06-06:
Yes, but the daily builds in the Play Store are currently out of date. Should be fixed in a day or so. After that, you can just install daily from play store and you'll get this.

DonLakeFlyer, 2018-06-27:
[image]

DonLakeFlyer, 2018-06-27:
That completes this work. Enter any new issues for other stuff which isn't working/mission.

closed 2018-06-27 by DonLakeFlyer

ChristopherOlson, 2018-06-27:
Don, this is nice! I studied the code for the first part you did and gained a basic understanding of how this is done for a GUI. I'll start playing with this and try it all out!

Thanks!