mavlink/qgroundcontrol #6445: Planner: mission time/distance ignore hold time, and loops.

AndKe, 2018-05-13:
If a waypoint has a hold-time specified, this does not affect the flight time.
also:
if waypoints are repeated (jump-to-item x times)  , this does not affect flight time or distance.

DonLakeFlyer, 2018-05-19:
Nope hold time never worked. There is a ton of missing time calculation if you start to use things beyond the basic. I think it's more likely to have to fail time calc as oppposed to figuring out how to calc for every crazy mission item possibility.

AndKe, 2018-05-19:
It does not need to be that way :)  QGC could loop thru a missions items that contain navigation & flow and actually sum them up.  - but yes, that's more work.   
Maybe the presented data could at least display warning when encountering a mission item it can't handle, or is known to brake it.

DonLakeFlyer, 2018-05-19:
Nothing needs to be that way given infinite coding time :)

DonLakeFlyer, 2018-05-19:
@dagar This is another example where it would be awesome for the firmware code (or something) to be somehow responsible for calcing flight times. Since that code is the only thing which really nows flight characteritics.

DonLakeFlyer, 2018-05-19:
Similar (sideways sort of) to our landing pattern distance calc thing.

dagar, 2018-05-19:
Good to know. Should we keep a list somewhere?

DonLakeFlyer, 2018-06-20:
Hold/Delay work now. Jump doesn't.

AndKe, 2018-06-21:
great, that's good news for payload that require hovering, it simplifies such missions.

IeiuniumLux, 2018-09-03:
@DonLakeFlyer, what do you mean with Jump doesn't work?  It's broken in QGC v3.4.2 or in the PX4 firmware v1.8.0?

DonLakeFlyer, 2018-09-03:
If you have a jump in your mission QGC will not calculate the time/distance correctly.