mavlink/qgroundcontrol #6446: It is unclear how "start mission" is supposed to work.

AndKe, 2018-05-13:
With a mission starting with takeoff (arducopter):

The "start mission", yellow box is visible  - I slide it, mode changes to AUTO, and I get
MAVLink msg: COMMAND_ACK {command : 300, result : 4}
- nothing happens.

If I click the "takeoff" button, in the fly menu...  - and slide on the yellow "takeoff" box,  the RPAS goes into guided, and takes off..

DonLakeFlyer, 2018-05-17:
What firmware version. This used to work. Are you trying to do it before you have GPS Lock?

AndKe, 2018-05-17:
AC 3.5.5 - does not work in real life 
If it was as simple as preflight checks, I would know :)
Also reproducible on current master in SITL

BTW: if it was as simple as GPS fix, it would not takeoff in GUIDED as well (the "takeoff" button works.)

DonLakeFlyer, 2018-05-18:
Sigh! Keeping up with ArduPilot breaking QGC is a full time job.

AndKe, 2018-05-18:
I guess it's a foolproofing attempt from the devs.  It's futile to foolproof, nature will just create a better idiot anyway.

closed 2018-06-01 by DonLakeFlyer