mavlink/qgroundcontrol #6448: Values widget data embedded in video stream record feature request

Patnetpatnet, 2018-05-14:
Date, time, lat, long, bearing would be really useful in video playback. Similar to how a camera time-stamps maybe.

jaxxzer, 2018-12-19:
This will require re-encoding the video which has significant hardware/resource requirements. I think setting up some process to format/export the video or subtitles in post will be the best route.

Patnetpatnet, 2018-12-19:
Hi, thanks for considering the question. The idea behind the request would be to provide a record of the flight with its visual realtime location with the video as a formal legal record. Better to understand than log and maps superimposition, now existing, for example. Would be good for quick replay in the theater too, in Search and Rescue, for example

dogmaphobic, 2018-12-19:
Log replay in sync with playing the recorded video. A lot of work but not impossible.

jaxxzer, 2019-06-05:
I think https://github.com/mavlink/qgroundcontrol/pull/7482 covers this.

jaxxzer, 2019-06-21:
I think this can be closed @DonLakeFlyer @dogmaphobic

closed 2019-06-21 by DonLakeFlyer