mavlink/qgroundcontrol #6449: change to "wp altitude" does not affect displayed waypoints

AndKe, 2018-05-14:
Attached video shows the issue perfectly.

[qgc-alt-change.zip](https://github.com/mavlink/qgroundcontrol/files/2002446/qgc-alt-change.zip)

In short: 
Using "waypoint alt" in "mission start" mission item , offers to change all altitudes in a mission.
but none are visibly changed, not until the mission is uploaded/downloaded.

AndKe, 2018-05-14:
Another important inconsistency that may cause similar problems:

those changed, (not visually changed) waypoints are correctly uploaded to RPAS, then if downloaded, good.
however: if user is working with files, not RPAS, then saving those changes to file, the loading file, does not help, the updated waypoints are not stored in a file..  nasty.  - are there two ways of fetching the same data ?

AndKe, 2018-05-15:
that was 7495626c3 from 2018-05-04 , - it's fixed in current master.   I did not see it in the RSS feed of commits since I wrote it down. sorry.

closed 2018-05-15 by AndKe

DonLakeFlyer, 2018-05-15:
No problem. Better safe than sorry.