mavlink/qgroundcontrol #6451: Feature Request: Modified Survey Mode

jstrebel, 2018-05-15:
We regularly fly a number of agricultural crops with a multi-spectral camera. 
(speed approx 13-15m/sec) Spacing approx 20m.
In missions with relatively moderate winds, we encounter the limits of flight plan control on a regular basis, at the turns at the end of the field. In order to reach the trajectory during the return flight with sufficient precision, we must, depending on the wind the mission at the ends by hand. In certain places, and depending on wind force, we can adjust the turnaround distance to solve the problem. However, there are often obstacles that make this impossible. We believe that a modified survey mode would help make it easier and more efficient to use.

To illustrate the flight mission I enclose a picture with the waypoints.
Actual survey mode: UAV flies 1,2,3,4,5 --- to mission end sequentially.
Desired “Modified survey mode”:
UAV flies points: 1,2,3, 180°radius turn to 30, heading toward 31, 180°radius turn to 7,
heading toward 6, 180°radius turn to 35, heading toward 34, 180°radius turn to 10, and so on.

This feature would be extremely helpful for Fixed Wing UAV’s because of their turn radius
is determined by the type of the Airplane.

[image]

DonLakeFlyer, 2018-05-15:
> We believe that a modified survey mode would help make it easier and more efficient to use.

Modified how? The plan is to allow for surveys to fly alternate lines. NYI, but coming.

DonLakeFlyer, 2018-05-15:
Sorry just seeing your example, looking...

Kjkinney, 2018-05-18:
This is similar to this issue https://github.com/mavlink/qgroundcontrol/issues/5866 also posted here. @Antiheavy what are your thoughts? I feel like this would be incredibly beneficial for fixed wings.

DonLakeFlyer, 2018-05-18:
The issue with skipping an arbitrary number of transects (skip determined by turnig radius) is what you do when you get to the end and have to go back and do the ones you skipping. That turnaround gets complicated if you are trying to fly shortest path.

Antiheavy, 2018-05-18:
Zamboni mode!  +1 for this.  I like the idea of a simple "fly alternating lines" checkbox.  You're right, it usually gets inefficient if you try to skip N lines at a time.

DonLakeFlyer, 2018-05-18:
So do I because it's easy. But it won't work if your transects are still too close to each other.

sanderux, 2018-05-23:
+1 for this, really needed for tight grids as we have also encountered.

DonLakeFlyer, 2018-05-24:
The main question for alternating transects is what to do when you exit the first pass? The turnaround for that will be a single width wide instead of double width.

DonLakeFlyer, 2018-05-24:
I have it working (without any fancy turnaround problem solved). And realized it's going to need new Planning visuals as well, which is going to end up being way harder than the 5 minutes it took me to implement the grid generation!
[image]

Antiheavy, 2018-05-24:
Maybe the return lines could be a slightly different color than the outbound lines? So colors alternate every other line basically?

sanderux, 2018-05-24:
you could also fly it like this;

1 3 5 7 9
6 8 4 2

sanderux, 2018-05-24:
Or, in case of even number
1 3 5 7 9
6 10 8 4 2

DonLakeFlyer, 2018-05-26:
Does this work in all cases? Makes my head hurt a bit to think about it.
* Fly first forward skip pass. Last transect flown is _LTF_.
* Forward skip pass from _LTF_ - 3 to last transect
* Backwards skip pass from _LTF - 3 - 2 to first transect
Taking into account the various boundary conditions of really small number of transect surveys.

sanderux, 2018-05-27:
Well, it might be a bit overcomplecating things.
You could simply set an additional wp in the turnaround at the point entry point where the next line would begin if the survey wasnt done yet

sanderux, 2018-05-27:
So
1 3 5 7 9 (fly as if there were an 11)
8 6 4 2

DonLakeFlyer, 2018-05-27:
> You could simply set an additional wp in the turnaround at the point entry point where the next line would begin if the survey wasnt done yet

The problem with that is that it extends the survey boundary past where the user specified. There might be an obstruction there. Hence it's better to keep within the specified bounds.

Antiheavy, 2018-07-18:
Just reporting in to say I've been using the new "fly alternate transects" option in Survey and I think it works great for the purposes described in this issue post.

closed 2026-10-04 by github-actions[bot]