mavlink/qgroundcontrol #6457: FW Landing Pattern: include flare params for correct approach alt calc

tstastny, 2018-05-16:
PX4 firmware includes a flare in the landing approach curve. This shifts the approach entrance point (loiter tangent location) back a bit for a given approach altitude. One can either 1) make this calculation before hand and put in the correct landing distance to the fw landing pattern gui or 2) put something close and let PX4 tell how much to move down or over.

It may be nicer for users if there would be an additional param option for the "HVIRT" value -- this defines a virtual distance below the land point that the landing slope intersects with. Adding this param as an input arg would allow the calculation of the correct waypoint on QGC for PX4 firmware. 

I could make a PR with this change generally (I know where to add the calculation) -- but I am unsure how to make it firmware specific, as I guess ardupilot may not use the same calculation and thus wouldnt want to see this param there in the gui if they cannot use it. I'd be happy to have any other opinions on how this should be done.

DonLakeFlyer, 2018-05-18:
I don't quite understand what you are saying here or what you want. In QGC you specifiy the height and distance to landing. The the firmware fits a landing approach into that space using whatever mechanism it uses. If the height and/or distance make it unable to fit a landing approach it will fail the mission upload.

Given that what is wrong or needs changing?

tstastny, 2018-05-18:
Yes you can choose the correct distance and altitude - but this needs to be calculated separately on your own with whatever flare params. And I doubt most general users want to do this (or even know what calculations go into the flaring curve). -- further, it is a bit tedious, if you don't make the calculation, to upload a mission, and then wait for the mission check to fail and then try and change whatever the "move down or over" message recommends.

I was only saying that it would be nice if we do the calculation for people. But this would require one extra param (specific to px4) as an input argument to the fw landing pattern editor. Which of course also requires that the firmware being used needs to be detectable. The latter point, I am unsure if QGC does.

DonLakeFlyer, 2018-05-18:
Ok, I get it now. @dagar and I have been talking through how to do this. But I haven't gotten back around to it yet.

tstastny, 2018-05-20:
@DonLakeFlyer I could make a first cut of a PR to get things rolling (save you some time). At least for this specific one (the flare param). But as I said before - I would need someone to point me towards a way to detect the firmware that is connected. Otherwise - let me know if there is another way you and @dagar would like to approach this, and I how can help.

DonLakeFlyer, 2018-05-20:
Knowing the firmware is easy. The problem is how do you make this work when you are not connected to a vehicle. In that case you don't have parameter values.

tstastny, 2018-06-26:
For anyone else interested in this -- I started a WIP branch with how one may want this to work -- of course specifically for px4 users. If this is something you would be interested to push -- I would need quite some help, as I'm terrible at qml (note how hacky this implementation is). But the basic functionality is at least there to demonstrate. Notes on what was changed and why in the commit message.

https://github.com/tstastny/qgroundcontrol/tree/feature/fw_landing_flare_params

Note: this simply changes the glideslope option to the px4 specific implementation and requires one make sure the onboard params and param input args in the editor are the same. Still a lot to be desired -- e.g. I couldn't get the fly map to update the flare area, but I suppose that is something someone more experienced could help with.

@philipoe

DonLakeFlyer, 2018-06-26:
Can you instead include some screenshots and explanation of how it works?

ryanjAA, 2018-11-03:
anything ever happen with this?

DonLakeFlyer, 2020-11-20:
Don't think this will ever bump to the top of the list

closed 2020-11-20 by DonLakeFlyer