mavlink/qgroundcontrol #6476: Something Wrong with Multiple Joysticks Like Logitech X56

smafaith, 2018-05-21:
When posting bug reports, include the following information:
- Ground station operating system: Ubuntu 16.04 and Windows 10. 
- QGroundControl version and build: stable and self-built from source. 
- Autopilot board: Pixhawk 2. 
- Autopilot firmware: PX4 and ArduPilot, latest self-built form source.
- Exact steps to reproduce the problem. Starting from booting QGroundControl to when the problem occurs:
I'm trying to use Logitech X56 and Thrustmaster Rudder Pedal as joysticks on qgroundcontrol for remote manual control, but I'm not able to calibrate the three joysticks together. This may take 3 USB cables on the laptop, so the QGC recognizes them as three different sets of joysticks. So I'm only able to calibrate roll and pitch with X56 Stick, throttle with X56 Throttle, and yaw with Rudder Pedal. But it's not possible for united calibration, let alone united manual control. Besides, the Rudder Pedal is not able to start to calibrate for less than 4 channels.

DonLakeFlyer, 2018-05-21:
This type of setup is not supported. Only a single joystick is supported by QGC.

closed 2018-05-21 by DonLakeFlyer

smafaith, 2018-05-22:
Then will this type get supported later? Or how can I try to develop and self-build the QGC to support that? Thank you. @DonLakeFlyer

DonLakeFlyer, 2018-05-23:
I doubt support for this will get added any time soon. It would not get used much and add a ton of complexity to existing joystick setup. It wouldn't be something easy to implement. Joystick source files all start with Joystick in the name.

MuhammadBilal1, 2023-02-17:
@DonLakeFlyer Is there any update on this?
I got [Logitech G X56 H.O.T.A.S](https://www.logitechg.com/en-us/products/space/x56-space-flight-vr-simulator-controller.945-000058.html), and now I want to use this with my QGroundControl for MultiCopter. Is this possible to configure this?

I connected this with my laptop and QGC shows the Joystick tab, but only 1 thing is working at a time.