mavlink/qgroundcontrol #6480: Request: command result GUI interaction

dagar, 2018-05-21:
# Problem
When trying to start a mission, takeoff, or directly arm the vehicle is sent a Mavlink COMMAND_LONG (or INT), and responds with a COMMAND_ACK containing a MAV_RESULT.

Example: Takeoff

1. Press Takeoff button
    - takeoff slider appears
    - slide to right
    - [image]

2. Takeoff is rejected
    - Receive generic command rejected message (from the COMMAND_ACK?)
[image]
    - Receive takeoff specific takeoff error message (STATUSTEXT) [image]

# Request

Can we improve this? 

1. Instead of showing the generic MAV_RESULT message (eg "command temporarily rejected") it should be done graphically and tied together with whatever initiated the command.
    - for example the slider could stay on screen and show something like a red X if the command was rejected, possibly even staying in view to try again?
    - maybe the button itself could change color to indicate the status? For example you press the Takeoff button and it slowly pulses some color, until it gets the MAV_RESULT and changes to green or red indicating success.
   - I don't really know how to do this graphically, but I do know having separate text message errors like this is bad.

2. Assuming 1 is solvable could we then find a way to associate the descriptive STATUSTEXT error message with the COMMAND you just sent. Is this effectively what the events interface solves?

@DonLakeFlyer @dogmaphobic does this make sense? Any ideas? I'd be willing to help in some capacity, although I don't know what the right answer is with respect to UX.

DonLakeFlyer, 2018-05-21:
Send a text error back in the MAV_RESULT as well?

dagar, 2018-05-21:
An optional ACK with a text error message? I actually think a good chunk of the most common errors could be handled via Mavlink spec enums (and QGC localization). For the rest (and short term transition) knowing that a specific error message is in response to a command would really help tie things together.

Does the button command progress and result make any sense? I'd really like the gui interaction to "feel" connected to the vehicle.

hamishwillee, 2018-05-21:
FWIW

> Does the button command progress and result make any sense? I'd really like the gui interaction to "feel" connected to the vehicle.

Yes, it makes a lot more sense to me. I would still want to see the result as a message though.

dagar, 2018-05-22:
Which result? I'm not opposed to having some more detailed messages, but I don't think they should be lost in the console with all other messages if they're the direct result of user action.

hamishwillee, 2018-05-22:
@dagar What I mean is that I am saying Yes to this question - https://github.com/mavlink/qgroundcontrol/issues/6480#issuecomment-390768357

ie Even with the slider behaviour described I'd want to see that (same) generic MAV result message  "command temporarily rejected"

closed 2026-10-04 by github-actions[bot]