mavlink/qgroundcontrol #6484: offline maps do not render in flight plan view

dgorissen, 2018-05-22:
Running the current daily build in an area where there is no network connection and I do not know my flight plans in advance.

So I have to make them on the spot. Hence I thought I would download some offline maps. However, these offline tiles do not load in the flight plan screen which makes flight planning impossible :/

DonLakeFlyer, 2018-05-23:
Seems to work ok for me. @dogmaphobic Gus, what do you need to determine whether something is goinf wrong here. Is there some logging to turn on to verify?

> also, when planning a flight path (e.g., survey) QGC is stil downloading terrain even though the elevation data has been cached

Can you add a separate issue for that.

dogmaphobic, 2018-05-30:
The maps used within the Fly and Mission Plan View were divorced from the Map Manager a while back. Instead of using the Map Manager, only a subset of hardcoded providers and types are supported within those views. If you downloaded offline map tiles from something that is not specifically set within the Fly view, it won't match/show. That's the only thing I can think of that would prevent (offline) maps from showing.

DonLakeFlyer, 2018-06-06:
@dgorissen Did you see if this was because you had the wrong map type selected?

DonLakeFlyer, 2018-06-18:
@dgorissen ?

dgorissen, 2018-06-18:
Sorry for the delay. This was a problem we hit in the field but a while later we managed to get it to work. Though unclear exactly how (sorry), it may well have been the map type. Closing this issue, will re-open if I hit it again.

closed 2018-06-18 by dgorissen