mavlink/qgroundcontrol #6488: Automated Bootloader update

bkueng, 2018-05-23:
On the Firmware side we constantly hit the 1MB flash limit for the fmu-v2 target. The limit is due to a hw errata, which newer boards don't have. However, since even recent boards come from the factory with an old bootloader that lacks the detection for the errata, we need to flash the v2 target.

So the goal here is to upgrade the bootloader in a user-friendly way, so that the v3 target can be used, which can use the full 2MB flash.

Most of the things are already in place to do that. Here are the proposed steps for the upgrade process:
- QGC should prompt the user to upgrade the bootloader on affected boards (v2 HW and old BL version)
- QGC checks if there's an SD card on the vehicle. If not, prompt the user to insert one

From that point no user interaction is required anymore.

- QGC then grabs the BL file from a server
- QGC pushes the file to the SD card via FTP
- There should be a CRC, maybe as a separate file
- QGC flashes the (modified) v2 target: this can be master. Required changes are:
  - enable the `bl_update` command
  - add ROMFS logic to upgrade the BL when it's found on the SD card during boot. The file will be removed after the upgrade, and another reboot is done.
  - add the CRC check
- After flashing, a reboot is triggered as usual and during bootup the BL is updated.
- Then QGC can flash the v3 target

Does this make sense? Once we have worked out how to do it, I can start on the Firmware side.

@dogmaphobic Lorenz wrote that you will implement the QGC side. Is this ok?

thomasgubler, 2018-05-25:
➤ Gus Grubba commented:

“ QGC should prompt the user to upgrade the bootloader on affected boards (v2 HW and old BL version)”How do I know when this should be shown?

“ QGC checks if there's an SD card on the vehicle. If not, prompt the user to insert one”
How do I detect a microSD card loaded on the autopilot?

“ @dogmaphobic Lorenz wrote that you will implement the QGC side. Is this ok?”
The firmware updating code is a huge pile of spaghetti as it tries to handle everything under the sun. I would rather stay as far away from it as possible. If I am to do this, I would redo the whole thing instead of just adding more pasta to the already full bowl of it.

bkueng, 2018-05-25:
> How do I know when this should be shown?

It needs to check the bootloader version. As far as I know QGC already knows how to do that.

> How do I detect a microSD card loaded on the autopilot?

Not sure if there is a mavlink message already? Don't we have that already for the camera?

>  If I am to do this, I would redo the whole thing instead of just adding more pasta to the already full bowl of it.

That's fine with me, but I think it needs discussion with @DonLakeFlyer and @LorenzMeier

dagar, 2018-05-25:
@dogmaphobic if you're looking for even more scope creep we're probably finally in a good position to do the json manifest with the new CI system.

https://github.com/mavlink/qgroundcontrol/issues/2797

thomasgubler, 2018-05-29:
>  I am to do this, I would redo the whole thing instead of just adding more pasta to the already full bowl of it.

Makes a lot of sense in terms of clean engineering. What would be the difference in time required for development of one implementation vs the other?

dogmaphobic, 2018-06-05:
Ok, I'm now looking at replacing the firmware update code. Not the actual mechanics, but rather, they way it stores the various firmwares for the various boards. I want to make it based off a virtual class with a default implementation but allowing it to be reimplemented by plugins and whatnot. There should be a decoupling of the update code and the list and source of supported firmwares. The new class implementation can either have things statically defined (hard-coded) or preferably, dynamically loaded off some repository with current boards, firmwares, etc. The UI should also reflect this, which for the most part, it already does. The layer that does the actual bootload checks and flashing are mostly fine as they are.

With that said, this has probably nothing to do with the original issue, but I might as well do it all at once. Regarding the original issue, yes, QGC already knows the bootloader version. It even has some code already that handles what I believe is what this issue is to fix: 

https://github.com/mavlink/qgroundcontrol/blob/master/src/VehicleSetup/Bootloader.cc#L586-L591

• Where is the _QGC then grabs the BL file from a server_
• Where is the _QGC flashes the (modified) v2 target: this can be master._
• How do I find out if there is an SD card loaded? 

>Not sure if there is a mavlink message already? Don't we have that already for the camera?

Yes, there is a `MAV_CMD_REQUEST_STORAGE_INFORMATION`, which I could target the AP comp ID. If that's implemented on the AP side, we should be good to go.

I think the rest is fine, albeit a boatload of work :P

thomasgubler, 2018-06-05:
➤ Gus Grubba commented:

Is there a way to preserve Github formatting? That looks like a big blob of text :) If not, it might be better to just add a link to the pertinent comment instead of copying the raw, unformatted text.

bkueng, 2018-06-05:
> Yes, there is a MAV_CMD_REQUEST_STORAGE_INFORMATION, which I could target the AP comp ID. If that's implemented on the AP side, we should be good to go.

Here you go: https://github.com/PX4/Firmware/pull/9596

LorenzMeier, 2018-06-05:
@dogmaphobic Would be awesome to display SD card status

bkueng, 2018-09-07:
I implemented a version on the Firmware side that does not depend on QGC changes: https://github.com/PX4/Firmware/pull/10426.

However this could still be useful: either as described above replacing the FW-based method, or QGC just prompts the user, sets the parameter `SYS_BL_UPDATE` and then reboots the vehicle.

LorenzMeier, 2018-11-29:
@dogmaphobic @DonLakeFlyer This has really now some level of urgency.

closed 2026-10-04 by github-actions[bot]