mavlink/qgroundcontrol #6489: Weird flight path after editing an imported KML

dogmaphobic, 2018-05-23:
Changing an imported KML causes some funky paths:

[image]

[image]

DonLakeFlyer, 2018-05-23:
Hmm, I had though that was mostly working better now. Guess not. Can you attach a KML or .plan which fails?

thomasgubler, 2018-05-24:
[testpolygon.kml.zip](https://github.com/mavlink/qgroundcontrol/files/2034353/testpolygon.kml.zip)

Here is a zip of the kml file. (github does not support uploading kml files apparently)

DonLakeFlyer, 2018-06-01:
From master:
[image]

This is correct. Could be specific to your plan survey settings. Attach a plan file which fails.

DonLakeFlyer, 2018-06-18:
Plan file?

DonLakeFlyer, 2018-06-20:
Reopen is you can get master to no work like I've shown above.

closed 2018-06-20 by DonLakeFlyer