mavlink/qgroundcontrol #6493: Planning: offer to apply "mission start" altitude to all mission items.

AndKe, 2018-05-23:
If user change "Waypoint Alt"   - then he is offered to apply it to all waypoints.
- please add same feature to "flight speed" 

[image]

Antiheavy, 2018-05-25:
This seems like a useful idea. Another method might be some sort of “select all” feature where altitude and speed can then be globally set to all waypoints with some important exceptions such as Land waypoints.

DonLakeFlyer, 2018-05-25:
I'm not convinced since the concept of what you are doing is kind of different. Waypoints all have altitudes in them. The tendency to set a mission all a the same altitude is common. Hence changing all the altitudes at once make sense to be able to adjust. But if you specifically change flight speed at various points in a mission. Then if you are adding specific new commands to the mission to change speed. If you don't want to do that anymore then it would make more sense to remove them. Otherwise you are just leaving a pile of flight speed change commands in the mission which turns into noops. That seems strange to me.

AndKe, 2018-05-25:
@DonLakeFlyer I understand your view perfectly :) 
I did not explain my use: 
What actually was the case, was a stereocam flight about 20 waypoints, at different speed/altitude.
It's easier to modify all to  5m/s , then half of them to 6m/s  , than to modify all from 3m/s to 4m/s and 4m/s to 5m/s.  -half the job :)

Which bring me to the minor-programming request.  (not just code reuse like this was)

- could we have a "offset altitude" / "offset speed" button ?
if user entered -10m , all waypoints(and splines) would be decreased by 10m.
, same for speed. 
That would be useful for reflying missions on lower/higher altitude or different speed.

closed 2026-10-04 by github-actions[bot]