mavlink/qgroundcontrol #6503: Prepare for AC3.6

AndKe, 2018-05-25:
Current release candidate produces a lot of errors, it may be time to adapt to AC3.6 ..
These errors (the yellow box)  should be logged to console, or somewhere, or be possible to recall later on.

[image]
[image]
[image]

DonLakeFlyer, 2018-05-25:
Thanks for checking this @AndKe. It's hard for me to keep up with multiple firmwares. There are a bunch of parameter name changes I need to put into the parameter name remapping system.

kikislater, 2018-05-26:
Concern Rover 3.3 and Plane 3.9 also : https://github.com/ArduPilot/MissionPlanner/issues/1843

DonLakeFlyer, 2018-05-26:
Coming in another pull

DonLakeFlyer, 2018-06-22:
I think I finally have this finished up!

closed 2018-06-22 by DonLakeFlyer