mavlink/qgroundcontrol #6505: Segfault ... network

AndKe, 2018-05-25:
Happened again when trying to use QT in the field, where I accessed full parameter list.
- Just clicking around, reproduces this fault..   - usually faster when in field, with real RPAS :)
See here: https://youtu.be/m8WBMLO9U80

```
(gdb) where
#0  0x00007ffff3a6548b in QNetworkProxy::QNetworkProxy(QNetworkProxy const&) ()
    at /home/andre/Qt/5.9.4/gcc_64/lib/libQt5Network.so.5
#1  0x00007ffff39e5e14 in QNetworkAccessManager::proxy() const ()
    at /home/andre/Qt/5.9.4/gcc_64/lib/libQt5Network.so.5
#2  0x00005555555f3aeb in QGeoTiledMapReplyQGC::cacheError(QGCMapTask::TaskType, QString) (this=0x55555e905b50, type=<optimized out>)
    at /home/andre/prog/qgroundcontrol/src/QtLocationPlugin/QGeoMapReplyQGC.cpp:185
#3  0x00005555555e26c3 in call (arg=<optimized out>, o=<optimized out>, f=<optimized out>) at /home/andre/Qt/5.9.4/gcc_64/include/QtCore/qobjectdefs_impl.h:136
#4  0x00005555555e26c3 in QtPrivate::FunctionPointer<void (QGeoTiledMapReplyQGC::*)(QGCMapTask::TaskType, QString)>::call<QtPrivate::List<QGCMapTask::TaskType, QString>, void>(void (QGeoTiledMapReplyQGC::*)(QGCMapTask::TaskType, QString), QGeoTiledMapReplyQGC*, void**) ()
    at /home/andre/Qt/5.9.4/gcc_64/include/QtCore/qobjectdefs_impl.h:169
#5  0x00005555555e26c3 in QtPrivate::QSlotObject<void (QGeoTiledMapReplyQGC::*)(QGCMapTask::TaskType, QString), QtPrivate::List<QGCMapTask::TaskType, QString>, void>::impl(int, QtPrivate::QSlotObjectBase*, QObject*, void**, bool*) (which=<optimized out>, this_=<optimized out>, r=<optimized out>, a=<optimized out>, ret=<optimized out>, ret=<optimized out>, a=<optimized out>, this_=<optimized out>)
    at /home/andre/Qt/5.9.4/gcc_64/include/QtCore/qobject_impl.h:120
#6  0x00007ffff2a16e71 in QObject::event(QEvent*) ()
    at /home/andre/Qt/5.9.4/gcc_64/lib/libQt5Core.so.5
---Type <return> to continue, or q <return> to quit---
#7  0x00007ffff51877ec in QApplicationPrivate::notify_helper(QObject*, QEvent*) () at /home/andre/Qt/5.9.4/gcc_64/lib/libQt5Widgets.so.5
#8  0x00007ffff518ec17 in QApplication::notify(QObject*, QEvent*) ()
    at /home/andre/Qt/5.9.4/gcc_64/lib/libQt5Widgets.so.5
#9  0x00007ffff29eb8c8 in QCoreApplication::notifyInternal2(QObject*, QEvent*) () at /home/andre/Qt/5.9.4/gcc_64/lib/libQt5Core.so.5
#10 0x00007ffff29edf5b in QCoreApplicationPrivate::sendPostedEvents(QObject*, int, QThreadData*) () at /home/andre/Qt/5.9.4/gcc_64/lib/libQt5Core.so.5
#11 0x00007ffff2a3ebb3 in  ()
    at /home/andre/Qt/5.9.4/gcc_64/lib/libQt5Core.so.5
#12 0x00007ffff6950287 in g_main_context_dispatch ()
    at /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0
#13 0x00007ffff69504c0 in  () at /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0
#14 0x00007ffff695054c in g_main_context_iteration ()
    at /usr/lib/x86_64-linux-gnu/libglib-2.0.so.0
#15 0x00007ffff2a3e1ff in QEventDispatcherGlib::processEvents(QFlags<QEventLoop::ProcessEventsFlag>) () at /home/andre/Qt/5.9.4/gcc_64/lib/libQt5Core.so.5
#16 0x00007ffff29e9eba in QEventLoop::exec(QFlags<QEventLoop::ProcessEventsFlag>) () at /home/andre/Qt/5.9.4/gcc_64/lib/libQt5Core.so.5
#17 0x00007ffff29f27f4 in QCoreApplication::exec() ()
    at /home/andre/Qt/5.9.4/gcc_64/lib/libQt5Core.so.5
#18 0x00005555555da05b in main (argc=<optimized out>, argv=<optimized out>)
    at /home/andre/prog/qgroundcontrol/src/main.cc:253
(gdb) list
90	
91	    return JNI_VERSION_1_6;
92	}
93	#endif
94	
95	/**
96	 * @brief Starts the application
97	 *
98	 * @param argc Number of commandline arguments
99	 * @param argv Commandline arguments
(gdb) 
100	 * @return exit code, 0 for normal exit and !=0 for error cases
101	 */
102	
103	int main(int argc, char *argv[])
104	{
105	#ifndef __mobile__
106	    RunGuard guard("QGroundControlRunGuardKey");
107	    if (!guard.tryToRun()) {
108	        return 0;
109	    }
(gdb) 
110	#endif
111	
112	#ifdef Q_OS_UNIX
113	    //Force writing to the console on UNIX/BSD devices
114	    if (!qEnvironmentVariableIsSet("QT_LOGGING_TO_CONSOLE"))
115	        qputenv("QT_LOGGING_TO_CONSOLE", "1");
116	#endif
117	
118	    // install the message handler
[... 138 lines cut ...]
244	                break;
245	            }
246	        }
247	    } else
248	#endif
249	    {
(gdb) 
250	        if (!app->_initForNormalAppBoot()) {
251	            return -1;
252	        }
253	        exitCode = app->exec();
254	    }
255	
256	    app->_shutdown();
257	    delete app;
258	    //-- Shutdown Cache System
259	    destroyMapEngine();
(gdb) 
260	
261	    qDebug() << "After app delete";
262	
263	    return exitCode;
264	}

```

DonLakeFlyer, 2018-05-25:
That actually seems like a regular old crash. Different than what I had fixed. @dogmaphobic an you take a look?

dogmaphobic, 2018-05-25:
I just happen to be testing Linux, though with Qt 5.11. I'll take a look...

AndKe, 2018-05-25:
@dogmaphobic As you see, I experienced it most often when connected to vehicle, *maybe* because one is clicking more around then.
Should you have any trouble reproducing, I would be happy to try with 5.11 - but please have in mind it's kind of random, and may take some patience to reproduce. - That is - unless you have an airborne RPAS, then it will happen at any inconvenient moment :)

dogmaphobic, 2018-05-26:
It might be one of those... I clicked on just about everything for a good 15 minutes and nothing went wrong. Yes, I was connected to a vehicle. I even took off and landed. The crash looks to be network related and as a result of some map tile fetching. I did a lot of things that would cause tons of tile fetching and switching back and forth. It never crashed.

All this was on current master using Qt 5.11

AndKe, 2018-05-26:
I've tried to build it on qt5.11.0 , both donlakeflyer/Arducopter3.6 , and current master, and I get:

```

Copying files
cp: cannot stat '/home/andre/Qt/5.11.0/gcc_64/lib/libQt5MultimediaQuick_p.so.5': No such file or directory
Makefile:2047: recipe for target 'release/QGroundControl' failed
make: *** [release/QGroundControl] Error 1
andre@loke:~/Downloads/qgroundcontrol-ArduCopter3.6/qgroundcontrol/build$ ls /home/andre/Qt/5.11.0/gcc_64/lib/libQt5MultimediaQuick*
/home/andre/Qt/5.11.0/gcc_64/lib/libQt5MultimediaQuick.la    /home/andre/Qt/5.11.0/gcc_64/lib/libQt5MultimediaQuick.so.5.11
/home/andre/Qt/5.11.0/gcc_64/lib/libQt5MultimediaQuick.prl   /home/andre/Qt/5.11.0/gcc_64/lib/libQt5MultimediaQuick.so.5.11.0
/home/andre/Qt/5.11.0/gcc_64/lib/libQt5MultimediaQuick.so    /home/andre/Qt/5.11.0/gcc_64/lib/libQt5MultimediaQuick.so.5.11.0.debug
/home/andre/Qt/5.11.0/gcc_64/lib/libQt5MultimediaQuick.so.5

```
what's with the "_p" ? - looks like a typo to me .

DonLakeFlyer, 2018-05-26:
Did you delete your build directory to try to fix?

AndKe, 2018-05-26:
tried that now, - no success, I simply don't have that file.   
There was a very similar issue few years ago:  https://github.com/mavlink/qgroundcontrol/issues/3998  , @dogmaphobic  fixed it then ?

AndKe, 2018-05-26:
@dogmaphobic true, I cannot reproduce the crash. 
- but can I trust it to fully work ? - I were afraid that this file may be required on a later time, during execution.

dogmaphobic, 2018-05-27:
> but can I trust it to fully work ? - I were afraid that this file may be required on a later time, during execution.

I don't know who or what is looking for that .so. The Multimedia library is used to display videos off the disk (nothing to do with streaming videos). I can't think of a place in QGC where that's used.

> the .sh is missing (not copied to "release") - which makes me fear other, critical, parts may be gone ?

What shell script?

> is 5.11.0 far from being approved for QGC ? - I did not see any problems with it, are you aware of any ?

It depends on what you consider "far". I just went through a first round of tests. Even though I did not see any problems, it's one sample. We need a bit of testing before calling it "official". For instance, I don't have anything Aurdupilot handy to test with. All my tests were on PX4 autopilots.

AndKe, 2018-05-27:
qgroundcontrol-start.sh  - I assume we should use it, even if it apparently works fine without..

AndKe, 2018-05-27:
I removed libQt5MultimediaQuick_p.so.5 from the Makefile (it was only copied)
Now: running the QT5.11 compiled version gives this interestig result:

running qgroundcontrol-start.sh   : presents empty application windows with menu, unable to close, have to terminate.
running ./QGroundControl : works as expected, even with extensive testing of features.

FWIW: apmplanner2 , compiled on qt5.11  - executed binary directly, works perfect, including the elevation widget.

```

andre@loke:~/prog/qgroundcontrol/build/release$ ./qgroundcontrol-start.sh 
inotify_add_watch("/home/andre/.config/ibus/bus/88068ace76ef430096168750f1405ff6-unix-1") failed: "No space left on device"
Settings location "/home/andre/.config/QGroundControl.org/QGroundControl.ini" Is writable?: true
Filter rules "qt.network.ssl.warning=false\n*Log.debug=false\n"
QGeoPositionInfoSourceGeoclueMaster(0x5597a575cea0)
MAVLinkLogManagerLog: MAVLink logs directory: "/home/andre/Documents/QGroundControl/Logs"
Map Cache in: "/home/andre/.cache/QGCMapCache300" / "qgcMapCache.db"
qrc:qml/MainWindowInner.qml:14:1: plugin cannot be loaded for module "QtPositioning": Cannot load library /home/andre/prog/qgroundcontrol/build/release/Qt/qml/QtPositioning/libdeclarative_positioning.so: (libQt5PositioningQuick.so.5: cannot open shared object file: No such file or directory)
qrc:qml/MainWindowHybrid.qml:25: TypeError: Cannot call method 'attemptWindowClose' of null
Killed

andre@loke:~/prog/qgroundcontrol/build/release$ ./QGroundControl 
inotify_add_watch("/home/andre/.config/ibus/bus/88068ace76ef430096168750f1405ff6-unix-1") failed: "No space left on device"
Settings location "/home/andre/.config/QGroundControl.org/QGroundControl.ini" Is writable?: true
Filter rules "qt.network.ssl.warning=false\n*Log.debug=false\n"
QGeoPositionInfoSourceGeoclueMaster(0x558bac68aa90)
MAVLinkLogManagerLog: MAVLink logs directory: "/home/andre/Documents/QGroundControl/Logs"
Map Cache in: "/home/andre/.cache/QGCMapCache300" / "qgcMapCache.db"
qml: _currentMissionIndex -1
Done
After app delete
```

dogmaphobic, 2018-06-05:
I assume the original crash, for which this issue was created is no longer there. The subsequent missing `Qt5MultimediaQuick_p.so.5` issue has been resolved elsewhere.

closed 2018-06-05 by dogmaphobic