mavlink/qgroundcontrol #6508: No option to track or center map on plane/copter?

Jmelz, 2018-05-26:
I installed qgc a few weeks ago, and was flying this past weekend. I am flying a fpv plane with an eagletree vector flight controller. It interfaces with a dragonlink receiver using open telemetry, which gets converted to a minimal mavlink stream. I noticed on the PC, I don't seem to have a moving map... Am I doing something wrong? I can't find a map center option to pan the map as the plane moves...

DonLakeFlyer, 2018-05-26:
Should happen automatically. What version of QGC on what OS?

Jmelz, 2018-05-26:
I'll double check the qgc version, but the is is windows 10.

And I remember this working before, however I think it stopped with my last qgc upgrade.

Android seems okay still

Jmelz, 2018-05-27:
Looks like Android is also not centering. PC is running 3.3.2 on win 10.

DonLakeFlyer, 2018-05-27:
When you say "centering" do you mean that the vehicle always stays exactly in the middle of the map? If so, that is not the way it works now. The map will only move when the vehicle moves outside the current visible area. Only then will it recenter to the vehicle position.

Jmelz, 2018-05-27:
Thanks! Yes, that is exactly what I meant. I had not ever let it get out of my sight, because I kept panning the map manually. I like this functionality as you describe it - because for short flights the map isn't panning all over - but I do wish it was an option that could be turned on or off. What is the threshold before it centers on vehicle position? 50 pixels, 100? or literally outside of the visible area?

DonLakeFlyer, 2018-05-27:
> I like this functionality as you describe it - because for short flights the map isn't panning all over

In general users like it working this way much better. It centers when it goes outside the visible area. Option is interesting, but would rather not add yet another option.

Jmelz, 2018-05-27:
Understood - I'll center my map on my 'in close' flying area, and that should do well for me.

As a side note, if declutter is the goal (which I agree with), perhaps an auto connect to a plane/copter on app opening? Maybe if there is only 1 craft? Would save a little time :)

DonLakeFlyer, 2018-05-27:
> perhaps an auto connect to a plane/copter on app opening?

YOu can set your link to autoconnect and it should work.

Jmelz, 2018-05-27:
Ah -  might be a bridge too far for me... I don't have a pixhawk, sik radio, px4 flow, librepilot, udp or rtk gps device.

My dragonlink shows up as a serial bluetooth link.

Jmelz, 2018-06-04:
Hrm, after using for a few flights, I still would like the option to constantly center on vehicle...

If a plane goes down, it is tough to see where it went down if it is at the edge of the screen. I think this should be an option in the menu, personally.

DonLakeFlyer, 2018-06-04:
Using the Center tool on the right you can center the map on the vehicle.

Jmelz, 2018-06-04:
Thanks Don - I am unfamiliar with that tool... but this would be a '1 time' center- not tracking the vehicle in the center I would assume... I am hoping to track vehicle in center via an option.

DonLakeFlyer, 2018-06-04:
One time center. I try not to add too many options to QGC since it just confuses users. So far I haven't gotten any other requests for going back to an option to always center. Hence I'd rather not add yet another thing to settings.

Jmelz, 2018-06-04:
Sure, understood - There is a community of FPV pilots using dragonlink as a radio system, who, until recently, were advised to use tower. 

I think QGC is better, due to offline maps, and many other features. This is an option that we've talked about in our community, so I'm just relaying information that not only I'd like to see, but some others would as well.

I understand your reluctance to add another option, but this is something I think has value for us that fly long distance missions. If a plane goes down, it's good to see the surrounding area at all times to help in locating it.

I changed my title to a request, just in case you change your mind.

Jmelz, 2018-06-06:
Flew again yesterday, and noticed that when the plane flies to the edges of the screen, there are periods of time that it is invisible because it is behind the widgets on the right, and the "takeoff/land/pause/play" controls on the left (remembering that I'm not arming the eagletree vector, as it's not possible).

I also don't see any way to click a button to center the display on PC or Android :(

DonLakeFlyer, 2018-06-06:
Recenter is fixed now. Sorry I was thinking wrong. There isn't a re-center button on the Fly View.

Jmelz, 2018-06-07:
Thanks Don... What does this mean exactly?

DonLakeFlyer, 2018-06-07:
It means the vehicle won't go under the tools on the left/right.

DonLakeFlyer, 2018-06-07:
Correct

Jmelz, 2018-06-07:
Thanks, I will try to grab daily build to test today.

Jmelz, 2018-06-07:
I guess since I didn't join the qgroundcontrol game until recently, I was unaware of the documentation:

https://docs.qgroundcontrol.com/en/FlyView/FlyView.html

`Fly Tools
On the left edge of the screen you will see the Fly Tools. The order of tools from top to bottom is:
Center map
Map Type
Zoom In/Out
Center Map
The Center Map tool allow you to center the map around various points such as home position, vehicle and so forth.
`

Kinda matches up with what I was thinking - "center map around vehicle... and so forth"

DonLakeFlyer, 2018-06-07:
Those docs are pretty out date with respect to where things are now. But should still be somewhat helpful.

Jmelz, 2018-06-07:
So for me, that tool doesn't appear (to center map)… but is it gone for good? or does it only show when vehicle is armed? In tower, you have a similar tool... you single press to center "for now" and press and hold to 'follow' vehicle. I was thinking that this was how it used to be for qgroundcontrol based on what you've said...

DonLakeFlyer, 2018-06-07:
It's been gone for a while. With the new automatic re-centering it isn't needed since the vehicle is always in view.

DonLakeFlyer, 2018-06-07:
Also should have thought about this a while ago. But if you want the vehicle to be forced to recenter you can just pan the map such that the vehicle is no longer in view. Wait a few seconds, and it will automatically pop back to the center.

Jmelz, 2018-06-07:
Sure, but it still won't stay centered...

This is what made me create this past

Jmelz, 2019-02-06:
So does this mean it's being planned for next minor version now?

PR #8173 (2019-12-25): Fixes
* Fix #6508. Add new Fly View: Keep Map Centered On Vehicle option
* Fix #7299. Allow reset to default for any firmware type
closed 2019-12-24 by DonLakeFlyer