mavlink/qgroundcontrol #6515: Preflight checklist concept and ideas

sanderux, 2018-05-27:
With reference to https://github.com/mavlink/qgroundcontrol/pull/6364
Notifying @DonLakeFlyer @dagar @philipoe @dogmaphobic @Antiheavy

Approaching this from a production scenario i have a few comments and would like to discuss some options on the current preflight checklist.

**Custom lists**
As every vehicle has its own specific checks there should be a way to customize the checklist. Ideally this information resides on the vehicle and is sent to the GCS upon connect, this could be a comparable structure as the mission protocol where it would also allow for modifying the default checklist from the GCS. Having it on the vehicle allows manufacturers to initialize the vehicle with the correct preflight checklist. The requirement of pre-flight checks should also be a vehicle specific setting, not a GCS one.

**Automated checks**
Several connectors could be written for automated checks. Some could be firmware supported features such as spinning motors and moving actuators, others could use webservices such as weather forecast and ADS-B. Absence of internet connectivity should be thought about in that case. Automated checks (if passed) should perhaps not consume real estate on the interface. It could show "Automated checks: pass" and only when failed show the ones that failed.

**Forcing checklist**
Currently a pre-flight check is a choice for the user. In certain production scenarios it would be desirable to force the checklist. I dont know if this should be enforced by the firmware too (you could launch a mission from a transmitter) but at minimum it should then be enforced by the GCS before arming is allowed.

**Storing checklist status**
As some checks require the user to look inside the GCS it should always be possible to exit the checklist without losing the status on checks.

I hope this gives some food for thought and i'm curious on your feedback/other ideas.

Antiheavy, 2018-05-27:
> information resides on the vehicle and is sent to the GCS upon connect,

interesting idea.  Some sort of file that lives on the SDcard maybe.  As you say like a mission file; or maybe like a start-up file or something in-between.

You can see manual our pre-flight checklist here: https://github.com/mavlink/qgroundcontrol/issues/4119#issuecomment-370298671 and as you can see it is mostly items than cannot be automated by the vehicle or the GCS.

philipoe, 2018-05-27:
I think the devs share many/most of your suggestions! The exact time frame on when these will be implemented is of course another question.

But would it make sense to shift/move this discussion back into https://github.com/mavlink/qgroundcontrol/issues/4119 in order to not have too many separate discussions/issues on this? Could one of the github-owners do this?

sanderux, 2018-05-27:
> The exact time frame on when these will be implemented is of course another question.

I think we will need to coordinate with devs on both sides. (PX4/QGC). I could take on some of the PX4 side stuff although dataman is not my expertise. 

> But would it make sense to shift/move this discussion back into #4119 in order to not have too many separate discussions/issues on this?

I choose to create a new issue as the other one is very long and has a resulting PR that is already merged. So consider this a discussion on proposed changes to an existing feature. We could even close the old one as the request has been addressed.

DonLakeFlyer, 2018-05-28:
> As every vehicle has its own specific checks there should be a way to customize the checklist. Ideally this information resides on the vehicle and is sent to the GCS upon connect, this could be a [...]

Just FYI: Custom builds of QGC allow you to fully customize your own pre-flight checklist. I understand that is not super helpful if you don't have a custom build.

Antiheavy, 2020-02-15:
@sanderux there has been lots of improvements to the preflight checklist recently in QGC.  Many of your proposals above are now working in some form.  

FYI @dakejahl

closed 2026-10-04 by github-actions[bot]