mavlink/qgroundcontrol #6534: "Waiting on Terrain Data for Server"

hitimo, 2018-05-31:


DonLakeFlyer, 2018-05-31:
If terrain data is working in other areas. My guess woule be that this area is missing data. If you can give me the lat/lon of a rect which covers the area I can ask the terrain folks if they have data there.

hitimo, 2018-05-31:
Thanks @DonLakeFlyer for your reply. The corners of the scan area are (UTM):
```
570688, 5174392
570045, 5174028
570854, 5172812
571306, 5173381
```

DonLakeFlyer, 2018-05-31:
Can you do this? Turn off Terrain and then save the Plan. Rename to .plan.txt and attach here.

hitimo, 2018-06-01:
Hi @DonLakeFlyer I attached the plan to my last post.

DonLakeFlyer, 2018-06-01:
@birchera Can you take a look at this? It fails to complete the TerrainPolyPathQuery send by Survey. It should be 48 segments, but after the 5th or so it needs to request a tile and then it never responds from that.

hitimo, 2018-06-04:
>  It should be 48 segments, but after the 5th or so it needs to request a tile and then it never responds from that.

@DonLakeFlyer @birchera Any ideas what could be the reason for that and how I could fix it? Thanks

birchera, 2018-06-05:
@DonLakeFlyer you added some more logging (https://github.com/mavlink/qgroundcontrol/pull/6542) did that reveal anything?

DonLakeFlyer, 2018-06-05:
Nope. That just told me that the PolyPathQuery never completes.

DonLakeFlyer, 2018-06-05:
@birchera Also if I change TerrainPathQuery to use TerringAirMapQuery instead of the offline system it works.

hitimo, 2018-06-06:
@DonLakeFlyer Could you please push the changes that made it work for you in a temp. branch? Thanks!

DonLakeFlyer, 2018-06-06:
Those changes are just for testing. QGC now always runs against the offline cache.

DonLakeFlyer, 2018-07-03:
Should be fixed now

closed 2018-07-03 by DonLakeFlyer