mavlink/qgroundcontrol #6535: New french regulations imacting ground stations

Dronotique, 2018-05-31:
Hi everybody,

Some new laws will appears in France in July .

One of them concerns the ground station.
This new law implies that the maximum flight altitude option could not be disabled and if you set a value more than 150m a warning message should appears.

I put you the "decret" and "arrete" in fench, if you are interrested by the subject.
https://old2017.federation-drone.org/wp-content/uploads/2015/01/5-Decret-limitation-capacite%CC%81-V03-08-2017.pdf

https://old2017.federation-drone.org/wp-content/uploads/2015/01/6-Arrete-limitation-de-capacite%CC%81-V03-08-2017.pdf

But I think the problem is more global.
Country specific laws should appears in a lot of country.
What is the better solution to implement them 

- Create a fork for each country?
- Implement them in master branch and activate them depending of your location?

Thanks

dogmaphobic, 2018-05-31:
I think it's unreasonable to expect QGC to handle 194 different countries and their capricious regulations. Just make the default something absurdly low (50m or something) and let users bump that up to whatever they want at their own risk, understanding that by doing so, they assume all responsibilities.

kikislater, 2018-06-02:
Futhermore law is still in project. Not voted, not official documents ...

closed 2018-06-02 by DonLakeFlyer