mavlink/qgroundcontrol #6540: Mission update: do not insist on pause while upload.

AndKe, 2018-06-01:
When a mission is updated while in progress, assume pilot is experienced enough to account for waypoint numbers or active waypoint and at least make "pause mission" optional.

Pausing a mission is rather disruptive to a mission update that do not need a pause.

DonLakeFlyer, 2018-06-01:
Commericial vehicle manufacturers would disagree with you quite strongly. Would rather keep as is without adding more "optionality" to it.

AndKe, 2018-06-01:
Is there a good reason for it ?
 (except; green pilot can't understand that some changes may be too big ?)

I have two reasons against it: 
Multirotor: trigging photos every x meters, at a given speed, gives pretty uniform stable pitch (assuming nose-ahead), when suddently going from 8m/s to breake, or accelerating again, there is a chance the next photo is trigged while the pitch is very ..unfortunate. (hard brake or acceleration)

Fixed wing: 
Switching circle(?) will not resume photos exact along the very same transect, it will most likely resume a bit to left/right of desired position, messing up horizontal overlap.

In addition; pausing mission when not needed will *always* spend extra energy doing nothing useful, on bigger missions, slower links , even more time.

having the operator hit brake when needed have no downside.  - having those stops/starts has always a downside.

Antiheavy, 2018-06-02:
Doesn’t QGC set the active waypoint to item 1 with Ardupilot?  I know this is what happens in Px4.

The firmware behavior you described for Ardupilot sounds decent enough.

AndKe, 2018-06-02:
no, I was able to modify (change some lateral positions) and upload w/o going back to WP1.

DonLakeFlyer, 2018-06-02:
@AndKe You are describing how ArduPilot works, whereas @Antiheavy is describing how PX4 works. When you upload a mission QGC does not set a current item in the mission. The firmware does what it thinks makes sense.

The issue is that QGC has no idea what the user did to the mission. In some cases it may make sense to continue from the previous waypoint. In other cases the previous wayopint may have no bearing at all. Also as you can see the behavior here is currently firmware dependant as to what happens if QGC sets not active waypoint. Because of that, QGC pauses the mission for safety and lets the user set the current waypoint manually as needed or not.

AndKe, 2018-06-02:
Yes, exactly. Because we can't expect QGC to understand the changes, all I ask for is to treat the operator as an sober adult :)  - do not override or take away PIC's control

Having only a button with option to "pause & upload" is ok, it may be a convenient option in some situations, not having it would require a mode switch.(simple woraround)

Not having a straight "upload" button takes away the option to perform mission as planned, uninterrupted, with perfect paths/velocity. There is no fix for that.

DonLakeFlyer, 2018-06-02:
I think the best thing to do here is add a setting to allow you to optionally not pause. The default for normal QGC would offer that, possibly not even showing the ability to change that setting in the UI to remove clutter. Then custom builds can still override that option to always force pause if they want.

AndKe, 2018-06-02:
a setting would be good. I do tend to suggest for others to use QGC for planning, so custom build does not solve everything, a GUI setting would be nice. 

I also think that the default should be as close as possible to what AP does, and what users are familiar with, making a convenience-feature like automatic pause the selectable option - but that's just me.  :)

Antiheavy, 2018-06-02:
It would be nice if PX4 had smarter mission index handling...  There are a number of well established behaviors/logic that could be followed (the Ardupilot example sounds like an okay one, although I'm not experienced with it).  I can't speak for the whole PX4 user community, but if PX4 didn't always reset to mission index 1 upon accepting a new mission, then I'd love to turn off the pause-then-upload button.