mavlink/qgroundcontrol #6541: UVC Device not working in android app

Jmelz, 2018-06-01:
I was under the impression UVC devices were working? I don't see the option in the android app.

dogmaphobic, 2018-06-01:
QGC doesn't do anything special to handle UVC devices. It simply uses Qt's own camera support:

```
QList<QCameraInfo> cameras = QCameraInfo::availableCameras();
```

If Qt can't see it, QGC won't see it either.

Jmelz, 2018-06-01:
QGC does seem to see it... as soon as I plug the UVC device in, QGC is shown as one of the apps to work with the device via a popup.

Specifically:
"select an application for the usb device."

fpviewer | go fpv | qgroundcontrol

Selecting qgroundcontrol does nothing...

dogmaphobic, 2018-06-01:
My guess is that the way QGC sets up the USB port (for serial connections), it's offered as a possible application for anything you insert into the USB port. I'll bet if you could plug a toaster into the USB port, Android would offer to start QGC.

With that said, it seems obvious the camera is recognized by Android itself. If that's the case, it would be a Qt issue (not using the proper API).

Jmelz, 2018-06-01:
All I can say is bummer... I have this nice video device that works but now I have to use some other app because it isn't really supported here...

Tower has the option, but tower seems to crash when the uvc option is checked... 

My only other option is to flip back and forth between qgc and fpviewer... not ideal, but I guess it's something.

dogmaphobic, 2018-06-01:
Looks like UVC for Android is low on their priorities: https://bugreports.qt.io/browse/QTBUG-58026

dogmaphobic, 2018-06-01:
By the way, this is a dupe. I knew this ringed a bell :) Closing in favor of the original (much better explained) issue: https://github.com/mavlink/qgroundcontrol/issues/4411

closed 2018-06-01 by dogmaphobic