mavlink/qgroundcontrol #6544: Daily build e8ec1f3c0 has blank white main window and will not exit

estechnical, 2018-06-02:
I just tried daily build e8ec1f3c0 and cannot make it exit. Rebooting the computer is the only way I appear to be able to close it.

It does connect to px4 autopilot, but never does anything more than this screenshot shows:

[image]

DonLakeFlyer, 2018-06-02:
What OS?

dogmaphobic, 2018-06-02:
Ubuntu 16.04. Some QML element failed to load. It would be on the console output explaining exactly what.

>Rebooting the computer is the only way

You can't kill the process? That sounds highly unlikely.

DonLakeFlyer, 2018-06-02:
Are you using an AppImage or are youd building yourself. If building yourself, first verify that AppImage is working. If not report here. If AppImage works, then something wrong with your build setup. Most likely Qt version.

DonLakeFlyer, 2018-06-02:
@dogmaphobic It's that library that was pulled out. Looks like it was needed!

```
qrc:///qml/QGroundControl/FlightDisplay/FlightDisplayView.qml:17:1: plugin cannot be loaded for module "QtMultimedia": Cannot load library /tmp/.mount_0BHsQt/Qt/qml/QtMultimedia/libdeclarative_multimedia.so: (libQt5MultimediaQuick_p.so.5: cannot open shared object file: No such file or directory)
```

DonLakeFlyer, 2018-06-03:
Which as far as I can tell is only needed by FlightDisplayViewUVC.qml which uses the Video control. Don't know much about how that all works.

dogmaphobic, 2018-06-03:
Odd, it run here. I will check...

DonLakeFlyer, 2018-06-03:
AppImage fails.

dogmaphobic, 2018-06-03:
I have no idea what AppImage is :P

dogmaphobic, 2018-06-03:
Got it. It looks like there are a few more things missing. `libQt5PositioningQuick.so.5` for example. Looking...

dogmaphobic, 2018-06-03:
By the way, when something like that happens, you can't "close" the application:

```
qrc:qml/MainWindowHybrid.qml:25: TypeError: Cannot call method 'attemptWindowClose' of null
```

We should add a check and if the object is null, skip the weird exit rigmarole.

dogmaphobic, 2018-06-03:
I think it's all there now. The old multimedia library changed names and there is a new requirement for QtPositioningQuick as well.

https://github.com/mavlink/qgroundcontrol/pull/6547

estechnical, 2018-06-03:
Sorry, I missed all this as I was sleeping... Yes, I was using the daily build appimage.

Thanks for looking into that so swiftly :)

dogmaphobic, 2018-06-05:
I'm assuming this is done.

closed 2018-06-05 by dogmaphobic