mavlink/qgroundcontrol #6545: "Emergency stop" is dangerously placed/designed

AndKe, 2018-06-02:
GUI offers a list of modes.
Right to the right of modes, is the "armed" which when clicked, display the same yellowish design with "drag to disarm"

It's quite possible, especially on touchscreens, or laptop mouse pad  to click "mode", (at this time the yellow "continue mission"  box appears)  - then accidentally press the "armed"   - which changes the box to disarm/terminate command.

Second scenario:  on upper part of map click for a guided destination, accidentally click "Armed" , and confirmation will cause crash.

Suggested design : 
Color difference, one big , prominently placed keyword, with otherwise smaller text.
example for disarming:  red frame box , bigger font says red "DISARM"  
example for mission continue box:  yellow box bigger font says "CONTINUE".

DonLakeFlyer, 2018-06-02:
Yeah, really good points here. Let me think about it a bit.

Antiheavy, 2018-06-02:
> Color difference,

I'm a big fan of using a different color, specifically RED for EMERGENCY STOP.

DonLakeFlyer, 2018-06-02:
I think two things:
* Color
* Emergency Stop should center slider on screen.

This way you get color change and position change which should clue the user in to the fact that they clicked something. Sliders are at bottom of screen to not get in the way during normal flight. But Emergency Stop is different, and is essense should get in the way, hence centered.

AndKe, 2018-06-02:
Good idea of different placement, except if it covers the RPAS on map (which may very well be in the middle of the screen if centered on vehicle)

The one time I did a flight termination, it was all about watching groundspeed and position on map, and estimating the impact area to precisely invoke termination at a spot with minimal chance of damage to third party.

closed 2018-06-05 by DonLakeFlyer

AndKe, 2018-06-06:
Looks great :)