mavlink/qgroundcontrol #6549: QGC incompatible with Android 8.1.0 on Pixel C

bradwilkinson, 2018-06-03:
OS: Android 8.1.0
Device: Google Pixel C tablet
QGC version: stable, latest on Google Play Store, June 2, 2018
Autopilot: n/a - couldn't get QGC

I tried to install the QGC app for Android from the Google Play Store, but there was a message: "Your device isn't compatible with this version." The Pixel C is always updated with the latest Android system, but the device itself is a couple of years old. The QGC app works great on my new Pixel 2 phone.

Here is a shot of the tablet's info:
[image]

bradwilkinson, 2018-06-03:
After posting this, I think I have answered my own question and figured out the issue - the PixelC doesn't have a GPS receiver! I run Bluetooth GPS, getting the output from my phone, and this works great for geology ion with mapping in the field. But it isn't native GPS on the tablet. Interestingly, the Tower app installs and functions normally using WiFi location, and the Bluetooth GPS.

DonLakeFlyer, 2018-06-03:
The app manifest doesn't regiter itself correctly for supporting devices with no GPS. I'll fix that.

DonLakeFlyer, 2018-06-05:
Can you try a latest daily build. Should not say it is not compatible any more.

bradwilkinson, 2018-06-05:
@DonLakeFlyer , I just tried the daily build on Google Play at 18:00EDT. It still says not compatible. The date there is April 7, 2017. Is there a different way or location for the daily build?

DonLakeFlyer, 2018-06-05:
Install from here: https://play.google.com/store/apps/details?id=org.mavlink.qgroundcontrolbeta. That said, the daily builds seem to be trailing behind by a few days so soemthing is wrong.

DonLakeFlyer, 2018-06-05:
You can also get latest android daily build from here: https://s3-us-west-2.amazonaws.com/qgroundcontrol/builds/master/QGroundControl.apk

bradwilkinson, 2018-06-05:
Yes, that's what I tried before, same answer.

bradwilkinson, 2018-06-05:
Second link worked great! Installed with no issues. That's wonderful, thanks for your help with this!!

DonLakeFlyer, 2018-06-06:
Good news. We're fixing up the store stuff soon. So you should be able to go back to install from there hopefully in a couple days and then you'll get updates.

closed 2018-06-06 by DonLakeFlyer