mavlink/qgroundcontrol #6554: Is QGC already switching to Qt 5.11?

tstastny, 2018-06-04:
@dogmaphobic Currently the documentation says 5.9.3 "only". However with the most recent master, there are some missing libraries -- in particular these two recently added ones: [1c2ab59](https://github.com/mavlink/qgroundcontrol/commit/1c2ab59cacfaa7a7ae846b86fb30df0f92f7a5dc)  -- which I guess are not found because they weren't there in 5.9.3.

E.g. ```cp: cannot stat ‘/home/thomas/Qt/5.9.3/gcc_64/lib/libQt5MultimediaQuick.so.5’: No such file or directory```

Should I switch the 5.11, or comment these lines out: [L113](https://github.com/mavlink/qgroundcontrol/blob/master/QGCSetup.pri#L113), [L117](https://github.com/mavlink/qgroundcontrol/blob/master/QGCSetup.pri#L117), in the setup pri for now?

DonLakeFlyer, 2018-06-04:
We are in the middle of switching to 5.11. At this point everything is there except for Windows builds. So, yes you should switch to 5.11.

closed 2018-06-04 by DonLakeFlyer

DonLakeFlyer, 2018-06-05:
@hamishwillee Can you update dev docs to specify we are on Qt 5.11.0 now?

hamishwillee, 2018-06-05:
@DonLakeFlyer Are we on that version for ALL platforms? If not, what are the exceptions?

DonLakeFlyer, 2018-06-06:
Yup, everything has moved to Qt 5.11.0 now

hamishwillee, 2018-06-06:
Thanks, done in https://github.com/mavlink/qgc-dev-guide/commit/0e1f04766a37b4d3834f9463fccff624e3fed512 (and already rebuilt here: https://dev.qgroundcontrol.com/en/getting_started/)