mavlink/qgroundcontrol #6566: Expose airmode parameter

RomanBapst, 2018-06-06:
@DonLakeFlyer @dogmaphobic We have been supporting 'airmode' in PX4 master for a while now and it will also be part of the next PX4 release.
Airmode can be activated via the parameter MC_AIRMODE (0 = off, 1 = ON).
When activated the output mixer basically is given more freedom to choose the motor signals which results in great performance increase. Especially, for racing platforms this feature is an absolute must but generally every platform including VTOL can greatly benefit from it.

**However**, there can be cases e.g. during initial rate tuning, when you don't want to have this mode active. There is a chance that an overly tuned system can get into a state where it ends up increasing it's height although the user commands zero throttle.
Therefore, at this point we would like to default this mode to off but still expose it in QGC such that the user can activate it.
I suggest to put it into the tuning section (maybe in the form of a checkbox) and have a brief explanation about the use case and possible consequences. We should e.g. mention that this mode should not be used during tuning.

Feel free to ask me about anything you need!

@bkueng FYI. We are also talking about adding this to the PX4 documentation.

bkueng, 2018-06-06:
One more thing: we plan to extend the MC_AIRMODE param from 2 states (0 = OFF, 1 = ON) to 3 states (0 = OFF, 1 = ON for roll and pitch, 2 = ON for roll, pitch and yaw). So we should plan for a dropdown rather than a checkbox.

@MaEtUgR fyi

hamishwillee, 2018-06-07:
I haven't heard of this airmode thing. Presumably it requires documentation so users can understand what it actually does, and how to turn it on? It sounds like this is not a flight mode per-se (e.g. not like "return"). Is there any more information, or can you create it?

DonLakeFlyer, 2018-06-07:
Related to #6438

RomanBapst, 2018-06-11:
@hamishwillee It's not a flight mode, it's more like a very low level setting that holds for all flight modes.
As mentioned above we should definitely have it documented.

hamishwillee, 2018-06-18:
@RomanBapst THanks. @bkueng Has done a really good job of documenting this as part of the update MC PID tuning guide - you can see the WIP here: https://github.com/PX4/px4_user_guide/pull/256/files#diff-f66bfa3667a8fdc67d9b65cae082f3c5R213

DonLakeFlyer, 2018-06-20:
I think Tuning is going to need to move to 3.5

MaEtUgR, 2020-02-25:
I think the airmode setting should still get exposed with a small description and disclaimer.

In short it determines if a multicopter is allowed to increase the collective thrust in order to reach its roll, pitch or even yaw authority which leads to highly improved performance in corner cases but can increase danger e.g. if control loops are unstable for setup or tuning reasons.

junwoo091400, 2022-10-19:
This is already exposed in PID Tuning tab & explained in the docs: https://docs.px4.io/main/en/config_mc/pid_tuning_guide_multicopter.html#airmode-mixer-saturation

[image]

hamishwillee, 2022-10-20:
@junwoo091400 I think that ONLY enables it during tuning  - i.e. it isn't persistent.

junwoo091400, 2022-10-20:
> expose it in QGC such that the user can activate it.
> I suggest to put it into the tuning section (maybe in the form of a checkbox) and have a brief explanation about the use case and possible consequences. We should e.g. mention that this mode should not be used during tuning.

Since the description was quite dense, I actually missed the essence :smile:, yep the GUI for setting Airmode persistently isn't there yet.

hamishwillee, 2022-10-20:
And perhaps it doesn't even belong. Depends a bit on how narrow the use cases are in which it help.  Question for the experts. But would be nice if the tuning could identify problem cases and tell people "hey, you could benefit from this"/

MaEtUgR, 2022-10-26:
> I think that ONLY enables it during tuning - i.e. it isn't persistent.

@hamishwillee I think it is or should be persistent. Because you want to *disable* airmode during tuning which is also the default and then after verifying the tuning works as expected enable it. (Although I haven't tested.)

> Depends a bit on how narrow the use cases are in which it help.

Every time you give not much collective thrust but require more roll, pitch (or yaw) torque than the remaining differential thrust allows. It's very well explained here: https://docs.px4.io/main/en/config_mc/pid_tuning_guide_multicopter.html#airmode-mixer-saturation In practice this mostly happens:
- If you want to command less collective thrust in manual mode: 8% already at the limit for no airmode with low stick see https://docs.px4.io/main/en/advanced_config/parameter_reference.html#MPC_MANTHR_MIN
- If you want to allow the controller to command lower collective thrust: 12% pretty high see https://docs.px4.io/main/en/advanced_config/parameter_reference.html#MPC_THR_MIN
- If you have a vehicle with large efficient propellers that benefits heavily from lower collective thrust to not catch unwanted lift e.g. while braking.
- If you do planned or unplanned extreme flight maneuvers. planned: racer or industrial high performance tested e.g. Astro, unplanned: limited actuation and needs to catch itself from an uncontrolled situation.

closed 2026-10-04 by github-actions[bot]