mavlink/qgroundcontrol #6577: Crashing when switching to airframe configuration view

johnweber, 2018-06-07:
We are seeing a crash that happens when an autopilot is connected and switching to the airframe view in the settings.  It seems to be identical to this issue reported on github last year.

https://github.com/mavlink/qgroundcontrol/issues/4569

Also, worthy to note, sometimes when switching to the airframe view, it crashes. Other times, it the view switch seems to hang long enough that the link to the autopilot is lost and regained. It seems related.

We are running QGC master with Qt 5.9.4 on an i.MX6-based platform running an Ubuntu 16.04-based LXDE distribution (on X).

DonLakeFlyer, 2018-06-07:
Can you verify that the very latest daily build pre-built AppImage has the same problem? If you are running master from a few weeks ago you will see crashes which have been fixed.

johnweber, 2018-06-07:
This was built from master taken late last week, and built with Qt 5.9.4. I'm not sure there is an AppImage that we can run on the system? We are i.MX6-based.

dogmaphobic, 2018-06-07:
You need Qt 5.11.0

johnweber, 2018-06-07:
Is integration complete for Linux with 5.11.0?  We used 5.9.4 because we were under the impression that 5.11.0 was complete for Mac OS/Windows, work was still being done for Linux.

dogmaphobic, 2018-06-07:
Yes, Qt 5.11 is now implemented for all platforms. There were some specific crash issues with Linux that were fixed along with the move to 5.11.

closed 2026-10-04 by github-actions[bot]