mavlink/qgroundcontrol #6578: I can't get QGroundControl to work ever on my Mac...

npilcher, 2018-06-08:
Hi
I have tried multiple times to get QGC to work on my Mac. At present I am trying to install the latest 3.3.2 version on a 13" Mac Air running OS X 10.9.5. Would love to hear of any solutions. Each time I try to boot it it crashes half way through the boot.

I have tried uninstalling and installing earlier versions, with the same end result. Possible solutions much appreciated, as I don't really want to have to buy a new laptop just to run this one programme....

Thanks in advance

DonLakeFlyer, 2018-06-08:
First thing to check is if the Daily Build also crashes for you.

DonLakeFlyer, 2018-06-08:
@dogmaphobic Do you know if there is a way to get stack traces off the Mac?

dogmaphobic, 2018-06-09:
When something crashes on macOS, a stack trace is automatically generated and presented to the user. It would be a matter of saving that and sending over to take a look. With that said, 10.9 is 5 years old. I don't really have any idea what kind of support there is for something that old. Trying a daily build wouldn't hurt either.

npilcher, 2018-06-10:
I thought I had attached that to the original post. Here’s the error message and all the text with it.

Dr. Nicolas J. Pilcher
 
Founder & Executive Director
Marine Research Foundation
136 Lorong Pokok Seraya 2
Taman Khidmat
88450 Kota Kinabalu, Sabah
Malaysia
 
Telephone: ++ 60 88 244089
Fax: ++ 60 88 244089
Email: npilcher@mrf-asia.org
Website: http://www.mrf-asia.org
LinkedIn: https://www.linkedin.com/profile/view?id=40221189

This message contains confidential information and is intended only for the individual addressed in the text of the message. If you are not the named addressee you should not disseminate, distribute or copy this e-mail. You cannot use or forward any attachments in the email. Please notify the sender immediately by e-mail if you have received this e-mail by mistake and delete this e-mail from your system. Marine Research Foundation, Kota Kinabalu, Sabah. info@mrf-asia.org

npilcher, 2018-06-10:
/Users/nicolaspilcher/Documents/Projects/UAVs/Nick's QGC-Mac error log.docx

npilcher, 2018-06-10:
Not sure if that will be accessible. Here's the full response.

Process:         QGroundControl [6565]
Path:            /Applications/QGroundControl.app/Contents/MacOS/QGroundControl
Identifier:      org.qgroundcontrol.qgroundcontrol
Version:         3.3.2 (???)
Code Type:       X86-64 (Native)
Parent Process:  launchd [137]
Responsible:     QGroundControl [6565]
User ID:         501

Date/Time:       2018-06-08 11:06:47.221 +0800
OS Version:      Mac OS X 10.9.5 (13F1911)
Report Version:  11
Anonymous UUID:  368D14D1-7B59-2489-A2D5-AC6C87483AF5

Sleep/Wake UUID: 4E036AE1-80CC-4DC7-89BC-EED7204885C4

Crashed Thread:  0  Dispatch queue: com.apple.main-thread

Exception Type:  EXC_BAD_ACCESS (SIGSEGV)
Exception Codes: KERN_INVALID_ADDRESS at 0x0000000000000151

VM Regions Near 0x151:
--> 
    __TEXT                 00000001094cb000-0000000109e62000 [ 9820K] r-x/rwx SM=COW  /Applications/QGroundControl.app/Contents/MacOS/QGroundControl

Application Specific Information:
dyld: in dlsym()

Thread 0 Crashed:: Dispatch queue: com.apple.main-thread
0   com.apple.CoreGraphics        	0x00007fff8bacc2bd CGFontCopyTableForTag + 17
1   com.apple.CoreText            	0x00007fff970be7f8 TBaseFont::CopyTable(unsigned int) const + 200
2   com.apple.CoreText            	0x00007fff970d081c TBaseFont::CreateTraitsValuesPerFontInfo() const + 1140
3   com.apple.CoreText            	0x00007fff970d02af TBaseFont::CopyTraitsInternal() const + 87
4   com.apple.CoreText            	0x00007fff970cfe3d TBaseFont::CopyAttribute(unsigned long) const + 493
5   com.apple.CoreText            	0x00007fff970bdeb1 TDescriptor::CopyAttribute(__CFString const*) const + 175
6   com.apple.CoreText            	0x00007fff970bdddf CTFontDescriptorCopyAttribute + 99
7   libqcocoa.dylib               	0x000000010e1052f7 0x10e095000 + 459511
8   libqcocoa.dylib               	0x000000010e104bad 0x10e095000 + 457645
9   libqcocoa.dylib               	0x000000010e107f9e 0x10e095000 + 470942
10  org.qt-project.QtGui          	0x000000010b13c85f 0x10b053000 + 956511
11  org.qt-project.QtGui          	0x000000010b13c66f 0x10b053000 + 956015
12  org.qt-project.QtGui          	0x000000010b13cb03 QFontDatabase::addApplicationFont(QString const&) + 227
13  org.qgroundcontrol.qgroundcontrol	0x000000010967484c 0x1094cb000 + 1742924
14  org.qgroundcontrol.qgroundcontrol	0x000000010962cc98 0x1094cb000 + 1449112
15  org.qgroundcontrol.qgroundcontrol	0x00000001094d47f4 start + 52

Thread 1:: Dispatch queue: com.apple.libdispatch-manager
0   libsystem_kernel.dylib        	0x00007fff8d6a1662 kevent64 + 10
1   libdispatch.dylib             	0x00007fff95014421 _dispatch_mgr_invoke + 239
2   libdispatch.dylib             	0x00007fff95014136 _dispatch_mgr_thread + 52

Thread 2:
0   libsystem_kernel.dylib        	0x00007fff8d6a0e6a __workq_kernreturn + 10
1   libsystem_pthread.dylib       	0x00007fff8fce8f08 _pthread_wqthread + 330
2   libsystem_pthread.dylib       	0x00007fff8fcebfb9 start_wqthread + 13

Thread 3:
0   libsystem_kernel.dylib        	0x00007fff8d6a0e6a __workq_kernreturn + 10
1   libsystem_pthread.dylib       	0x00007fff8fce8f08 _pthread_wqthread + 330
2   libsystem_pthread.dylib       	0x00007fff8fcebfb9 start_wqthread + 13

Thread 4:
0   libsystem_kernel.dylib        	0x00007fff8d6a0e6a __workq_kernreturn + 10
1   libsystem_pthread.dylib       	0x00007fff8fce8f08 _pthread_wqthread + 330
2   libsystem_pthread.dylib       	0x00007fff8fcebfb9 start_wqthread + 13

Thread 5:
0   libsystem_kernel.dylib        	0x00007fff8d6a0e6a __workq_kernreturn + 10
[... 315 lines cut ...]

VM Region Summary:
ReadOnly portion of Libraries: Total=222.3M resident=82.5M(37%) swapped_out_or_unallocated=139.8M(63%)
Writable regions: Total=86.9M written=9188K(10%) resident=11.8M(14%) swapped_out=0K(0%) unallocated=75.1M(86%)
 
REGION TYPE                      VIRTUAL
===========                      =======
CG backing stores                   232K
CG image                              4K
CG shared images                    180K
Kernel Alloc Once                     8K
MALLOC                             58.2M
MALLOC (admin)                       32K
STACK GUARD                        56.0M
Stack                              11.6M
VM_ALLOCATE                        16.3M
__DATA                             23.6M
__IMAGE                             528K
__LINKEDIT                         74.7M
__TEXT                            147.7M
__UNICODE                           544K
mapped file                        47.9M
shared memory                         4K
===========                      =======
TOTAL                             437.5M

DonLakeFlyer, 2018-06-15:
Qt Bug: https://www.bing.com/search?q=qt+addApplicationFont+crash+on+osx&form=APMCS1&PC=APMC. No helpful information as to workaround.

DonLakeFlyer, 2018-06-20:
@dogmaphobic Any ideas here?

dogmaphobic, 2018-06-20:
My guess is something not compatible with macOS 10.9. I don't have anything that old to test with. With that said, there is an existing issue with (current) iOS that sounds similar to this (font related).

dogmaphobic, 2018-06-21:
In the process of looking into these issues, I found some issues with the macOS daily build. I do not know if your case specifically is related but let's see how this goes once it's fixed.

dogmaphobic, 2018-06-21:
By the way, the iOS issue I mentioned above seems to have gone away with the switch to Qt 5.11. It was a very similar issue as it was crashing at boot time due to some font related calls. Hopefully this will all get fixed at the same time.

DonLakeFlyer, 2018-07-13:
Ok, newer OSX is working fine. So I'm closing this.

closed 2018-07-13 by DonLakeFlyer