mavlink/qgroundcontrol #6583: Motors not registering on QGC

sjg110, 2018-06-11:
This system is for posting bugs or feature requests ONLY. For questions about how to use or build QGC see: http://qgroundcontrol.com/#resources

----
When posting bug reports, include the following informaiton:
- Ground station operating system.
- QGroundControl version and build (daily, stable, self-built from source, etc.)
- **If you are using a Stable build which is not the latest, first step is to install latest Stable build and try again prior to entering a bug report.**
- Autopilot board: Pixhawk I, Pixhawk Mini, Pixhawk 2, etc.
- Autopilot firmware (e.g. PX4, ArduPilot, custom) and version.
- Exact steps to reproduce the problem. Starting from booting QGroundControl to when the problem occurs.

**All this template text should be deleted before creating your issue.**

sjg110, 2018-06-11:
Hi I have the BlueROV2 set up and have downloaded the most recent QGC for Windows. I have calibrated the ROV from QGC, but motors is not appearing in the QGC. The motors are working and can be controlled from the wireless joystick.

Is there something I have missed? 
thanks

DonLakeFlyer, 2018-06-15:
@jaxxzer ?

sjg110, 2018-06-15:
I have found the motor controls in the parameter settings - This is just different to the instructions, so I first thought it may be a bug or something, but seems to be a reassignment. 
Thanks,

closed 2018-06-15 by sjg110