mavlink/qgroundcontrol #6585: When we plan this kind of flight, we fly a lot of useless routes.

MyOnlyOnlySunshine, 2018-06-12:
Hello，when we plan this kind of flight, we fly a lot of useless routes.  
How should we plan this graphic flight area so that we can not fly so many useless routes?
Thanks

[image]

kikislater, 2018-06-12:
Same as https://github.com/mavlink/qgroundcontrol/issues/6490
Is this a real flight over this kind of polygon ? More reasonable is like that : https://github.com/mavlink/qgroundcontrol/issues/6489

MyOnlyOnlySunshine, 2018-06-13:
Do you mean that we should cut into multiple convex polygons to fly? 
When we plan irregular navigation, it is easy to cause many routes to fly.

DonLakeFlyer, 2018-06-15:
First step is to use a newer daily build (assuming this is android) which will do a much better job at this. If iOs then you are kind of stuck for now until we figure out how to get working again.

DonLakeFlyer, 2018-06-18:
A similar pattern using daily build:
[image]

It's not going to get much better than that without some really sophisticated path generation which is unlikely to happen.

closed 2018-06-18 by DonLakeFlyer

MyOnlyOnlySunshine, 2018-06-19:
Could you help me find which C++ or qml files changes if i want to plan that routes ,i can not find the file of source code to learn about it . Thanks

MyOnlyOnlySunshine, 2018-06-20:
Amazing, too complex . Thanks