mavlink/qgroundcontrol #6602: "IGN MISSION_ITEM: No transfer" error

feemi, 2018-06-18:
An error occurs after uploading any mission to vehicle.

Which says "IGN MISSION_ITEM: No transfer". 
[image]

It seems like it sends mission correctly. When i reopen qgc it downloads mission from vehicle successfully.

Environment:
QGC v3.3.2 or QGC Master
PX4 1.7.4 Beta

Kjkinney, 2018-06-18:
We were getting the same issue and have found that it appears to be a firmware issue in px4. Try an older version of px4 or update to the px4 master and it should go away.

feemi, 2018-06-18:
Thanks for the feedback