mavlink/qgroundcontrol #6605: Assertion failed: (buffer->content_type == HB_BUFFER_CONTENT_TYPE_UNICODE), function hb_shape_full, file hb-shape.cc, line 358.

kenshinyao, 2018-06-19:
2018-06-19 22:52:51.232763+0800 QGroundControl[5346:2234591] [DYMTLInitPlatform] platform initialization successful
2018-06-19 22:52:51.785339+0800 QGroundControl[5346:2234399] Settings location "/var/mobile/Containers/Data/Application/99599456-1DA1-49C7-A117-D16E31C270E7/Library/Preferences/org.qgroundcontrol/QGroundControl.ini" Is writable?: true
2018-06-19 22:52:51.894919+0800 QGroundControl[5346:2234399] Filter rules "qt.network.ssl.warning=false\n*Log.debug=false\n"
Could not open log file '/var/mobile/Containers/Data/Application/C0DECEEC-108A-4262-A57E-E2C06F9D98E7/Documents/QGroundControl/Logs/gst/gstreamer-log.txt' for writing: No such file or directory

Assertion failed: (buffer->content_type == HB_BUFFER_CONTENT_TYPE_UNICODE), function hb_shape_full, file hb-shape.cc, line 358.
(lldb) 

ios
qgroundcontrol:master
xcode Version 9.4
qt 5.11.0

how can i fixe this bug?i can't build it in iphone.

DonLakeFlyer, 2018-06-20:
@dogmaphobic Is this the gstreamer problem you were having on iOS?

dogmaphobic, 2018-06-20:
It's the same issue I am having with iOS, but I'm not sure it's related to GStreamer. I need to exclude it and see if it makes a difference... 

This issue is quite similar to #6578 even though that one is macOS (an older, 10.9 version).

DonLakeFlyer, 2018-06-20:
Ok, so we can use this to track it then

dogmaphobic, 2020-02-11:
Ancient history

closed 2020-02-11 by dogmaphobic