mavlink/qgroundcontrol #6616: Cannot connect with MBP to inav 1.9.1

wx4cb, 2018-06-20:
Build: Latest build of qGroundControl available on website.

FC: FuriousFPV F35 - Inav 1.9.1

Receiver connection: Crossfire full module tx, micro receiver

Connection to FC: Bluetooth via crossfire.

What i'm experiencing on my 1.9.1 inav build with the F35 is I can connect fine using my android tablet using the mission planner app, and also occasionally with ezgui (although it's  not supported), But i cannot connect via the MBP. All I get is that it's connected in the comm link tab), but when I go to the main map screen (clicking the paper airplane icon) I'm just getting "waiting for vehicle connection".

I'm running qGroundControl on a Mac powerbook 13 inch (no touch bar) with high sierra through my crossfire bluetooth.

I have no problems doing that to get at the configurator via bluetooth through the module. and That's the same way I connect to the FC with Mission planner

micaheli, 2018-08-17:
I too am having this same issue with the windows version.  Works on Android, not on PC.

dogmaphobic, 2018-08-18:
I do, though I haven't used it in quite a while. I do remember having issues with macOS as well. I will take a look.

micaheli, 2018-08-18:
I can help test as well. I have a MBP as well as windows and Linux PC's.  I just prefer the windows PC to work since that's my field laptop/groundstation machine.

DonLakeFlyer, 2019-01-19:
@dogmaphobic Need to get this figured out before 3.5 goes out.

wx4cb, 2019-01-19:
Still available to test

closed 2026-10-04 by github-actions[bot]